yaapu/FrskyTelemetryScript #4: Info not always updating

kantlivelong, 2018-06-22:
I've noticed that the info displayed/played(audio) doesn't seem to update all the time. For example I switch the flight mode and the info is not displayed(or sound played) for minutes after. Is there a way to tell what the cause of this is?

TX: Taranis X9D+ (OpenTX 2.2.1 - lua,luac)
FC: PixHawk 2 (ArduCopter 3.5.5)

kantlivelong, 2018-06-24:
> Hi, it's just the audio or even the flight mode on the display is not
> updated!

So far I’ve noticed the flight mode, battery, and current information. I’d
imagine everything sent via telemetry is not being updated properly. I did
notice that the flight timer in the upper right hand corner is still
updating but without looking at the source I suspect that’s a local timer.

yaapu, 2018-06-24:
I've seen this issue with a defective or slow SD card, can you rule this out with a class 10 high performance card?

Usually as soon as you arm telemetry gets really slow

kantlivelong, 2018-06-24:
It’s the stock one that came with the radio. Luckily I do have a spare
class 10 from a camera. Will give it a shot and get back to you.

yaapu, 2018-06-24:
I meant on the pixhawk side, as soon as you arm ardupilot starts logging to the SD card and the frsky telemetry slows down or freezes, this is a known issues with less than perfect SD

kantlivelong, 2018-06-24:
Ah. Would that affect other devices that use telemetry? I have an OSD that communicates through MavLink and that is updating properly.

yaapu, 2018-06-24:
I've seen it on frsky only.
Does it work ok before arming and then after arming is less responsive?

kantlivelong, 2018-06-24:
It seems fine before arming and possibly for a short time after arming.

yaapu, 2018-06-24:
Mhmm, just to be sure would you switch card on the pixhawk

kantlivelong, 2018-06-24:
Will do. Hopefully sometime today but there is a chance of rain :(

kantlivelong, 2018-06-24:
Alright so I used a SanDisk 32GB Class 10 MicroSD (DSQUNC-032G-GN6MA) and the issue still occurs.

I did the following:
1. Tested switching between flightmodes before arming (worked fine).
2. Armed with Stabilize
3. Took off
4. After getting 15ft in the air switched to Loiter. At this point the radio did not indicate Loiter was activated and still showed Stabilize.
5. Switched to RTL. Still no change in display.
6. The copter landed and once landed the UI updated with RTL and that the landing was complete.

kantlivelong, 2018-06-24:
Additional info: I kept an eye on the FlightMode value in the settings window for the copter on OpenTX and the value did not update there either.

kantlivelong, 2018-06-24:
I set the LOG_BITMASK to 0 and now it's working which fits with your SD write speed theory. I just can't fathom how that SD isn't fast enough.

yaapu, 2018-06-24:
you're welcome, can I close the issue?

yaapu, 2018-06-24:
you're right I should add a trobleshooting section in the radme...

kantlivelong, 2018-06-24:
Closing.

Thanks for your help and writing this awesome script :+1:

closed 2018-06-24 by kantlivelong