yaapu/FrskyTelemetryScript #5: Master yaapu7.lua file is only 33.7kb...

LTMNO, 2018-07-15:
Hello Yaapu, it seems that the latest commit of the file has been saved incorrectly... the file is smaller than it should be and has wild characters within.  I have tried to use the History dating back a few months ago... no luck with those files either as it throws "Script Panic not enough memory" errors.

All works on the simulator though... this is for Taranis Q7+ running OpenTx 2.2

yaapu, 2018-07-15:
Hi what are you trying to achieve?

LTMNO, 2018-07-16:
Hello, I tried to execute the .lua and .luac scripts on my Taranis QX7+ when the files were loaded on the SD Card.
That was unsuccessful, then I created the .luac file via Companion as instructed in the readme.txt,
That didn't work either... 
I then manually created my own version of the file using the command line... that worked and then loaded it on the SD Card after compiling in Companion.... no luck there either... 

I either get errors in the script or memory error.

I did find an older version "https://github.com/yaapu/FrskyTelemetryScript" here and that was able to load on my radio.  its version 1.1 but from what i read, there is v1.6 available.

Hope that explains everything... 

I am not sure if something was wrong in the Lua Script.. i was going to do some file compariosn on the 1.1 version to the 1.6 version and see what has changed.

Great work btw, thanks for the contribution...

LTMNO, 2018-07-16:
Hello Alessandro, 
I had spent the better part of yesterday trying to get your solution up and running my Taranis QX7+
I am running OpenTx 2.2 and had loaded your latest Lua scrips posted…

Hello, I tried to execute the .lua and .luac scripts on my Taranis QX7+ when the files were loaded on the SD Card.
That was unsuccessful, then I created the .luac file via Companion as instructed in the readme.txt,
That didn't work either...
I then manually created my own version of the file using the command line... that worked and then loaded it on the SD Card after compiling in Companion.... no luck there either...

I either get errors in the script or memory error.

I did find an older version "https://github.com/yaapu/FrskyTelemetryScript <https://github.com/yaapu/FrskyTelemetryScript>" here and that was able to load on my radio. its version 1.1 but from what i read, there is v1.6 available.

Hope that explains everything...

I am not sure if something was wrong in the Lua Script.. i was going to do some file compariosn on the 1.1 version to the 1.6 version and see what has changed.

Great work btw, thanks for the contribution...

yaapu, 2018-07-16:
Hi,
there's nothing special to do to run the script on a QX7.
Just copy the /TARANIS/SD/ folder content to your radio SD card, make sure you have

/SCRIPTS/TELEMETRY/yaapu/copter.lua
/SCRIPTS/TELEMETRY/yaapu/rover.lua
/SCRIPTS/TELEMETRY/yaapu/plane.lua
/SCRIPTS/TELEMETRY/yaapu/copter.luac
/SCRIPTS/TELEMETRY/yaapu/rover.luac
/SCRIPTS/TELEMETRY/yaapu/plane.luac
/SCRIPTS/TELEMETRY/yaapu7.lua
/SCRIPTS/TELEMETRY/yaapu7.luac
/MODELS/yaapu
/SOUNDS/yaapu0/*

Then please run it as a telemetry script by setting it up in the model telemetry page (page 13 or 14), use screen1, select type script, choose yaapu7, exit and then from main model page long press page tu start it!

that's all required to run it

LTMNO, 2018-07-16:
Hello Alessandro, thanks for the reply....
I went ahead and formated the SD Card... I uploaded the information above... again.... this time removed the LUAC from the OpenTx Firmware options so just LUA enabled.
And it worked... I can see and hear that the script loads...

Unfortunately, I get the "No Telemetry" Banner in front of the Telem1 screen.

I have read in forums that the RxBt or RSSI is used to disable that.  I have captured my Telemetry screen... I have all the data showing in Telemetry Config Screens - all data is streaming.
One Caveat is that my Tmp1 and Tmp2 are Id 0002 and 0005 respectively instead of 0400 and 0410.  Also my Fuel Id is 0004 instead of 0600
and Finally, no VSpd is getting picked up from the Sensors... but that could be because it is not moving? 

Will that break the system?

yaapu, 2018-07-17:
You have the serial protocol set to 4 in mission planner, you have to set it to 10 delete all sensors and run discovery again. You need OpenTX 2.2 and a recent ardupulot

yaapu, 2018-07-17:
Did you manage to make it work?

I do have an account for donations, and they are very much appreciated, I use them to buy new equipment here's the link https://www.paypal.me/yaapu

LTMNO, 2018-07-17:
Okay, so I might have a problem… 
I am using Arducopter 3.2.1 on APM 2.7.2
I do have a PixHawk that I can swap this out for… but I was hoping to use my APM.
I am searching and apparently I don’t have that Parameter in my list… 
I am just researching when it was introduced.

Am I missing something?

yaapu, 2018-07-17:
Copter 3.2.1 is not supported, it does not have native passthru, but with a mavlink to passthru converter should work just fine, this is the link

https://github.com/zs6buj/MavlinkToPassthru

you'll also need a teensy to run it though

LTMNO, 2018-07-17:
Yes, thanks... currently I am using an Arduino with Converter Software written by MikeB. https://github.com/MikeBland/APM_Mavlink_to_FrSky
Its exposed 90+% of the data minus the anomalies I mentioned above.
I will try the other solution now... i have a teensy board as well.

LTMNO, 2018-07-17:
Thanks... just curious as I am trying to get the Teensy passthru working as well... but the ArduinoProMini solution with MikeB was working... just some Id's were off... could i not just adjust the Id's for your script to work?

yaapu, 2018-07-17:
MikeB uses a different protocol that has become obsolete with opentx 2.2

LTMNO, 2018-07-17:
Understood, thanks… I will reach out to MikeB and see what he thinks about updating it… he originally wrote it for er9x and ersky9x.

Again, amazing work!  Thanks...

LTMNO, 2018-07-18:
Hello Alex, troubles in paradise unfortunately....  The script executes on my QX7S and runs, I get Batt, Arming/Disarm, GPS Lock, working... audibles too... but nothing else.
I know I am receiving data as the No Telemetry Prompt is not there... I just did 2 flights this afternoon, and I was getting msg prompts from the APM in you notification screen.. like AutoTune started and Successful and all... so that was cool.

But nothing else is coming in?  Very Strange.
I am attaching the XLS Log... just remove the .xlsx from it to load.

[DART450-2018-07-18.csv.xlsx](https://github.com/yaapu/FrskyTelemetryScript/files/2207236/DART450-2018-07-18.csv.xlsx)

Also, the script won't load in Companion 2.2. anymore either, says script error?  I have tried to restore the originals with no avail.  Really Strange.

yaapu, 2018-07-22:
I agree with you that it is more related to apm 3.2.1 + eric's code.
All the testing that Eric has done with the help of many of us has been done on very recent ardupilot versions, your setup is something new for his project but should nevertheless work, mavlink is mavlink

yaapu, 2018-08-03:
Hi Pino,
I close it for you closed the linked issue https://github.com/zs6buj/MavlinkToPassthru/issues/5

closed 2018-08-03 by yaapu

LTMNO, 2018-08-03:
Thank you, I have not determined if the Mah Used or % used works yet... I am going to take a flight within the next 1hr or so.
I will report back.

Thanks Alex.

yaapu, 2018-08-04:
I get them from the teensy that in turn gets them from mavlink, do you see correct values in MP?