yaapu/FrskyTelemetryScript #7: Fuel or Mah Consumption not working?

LTMNO, 2018-08-20:
Hello Alex, 
I finally was able to get my Quad fixed... tore it apart and fixed the Power Module to be in line with power distribution vs passive install.  Long Story... ;-) . End of the day, I was am getting Curr now and unfortunately, the Consumption or Mah/Fuel is not being calculated?  See attached Log.
[DART450-2018-08-20.xlsx](https://github.com/yaapu/FrskyTelemetryScript/files/2304170/DART450-2018-08-20.xlsx)

yaapu, 2018-08-20:
Hi, the script does not calculate consumption it gets it from the flight controller so you should check why it's not transmitted, do you see consumption on mission planner?
If I recall right you are using the teensy, correct?

LTMNO, 2018-08-20:
Oh, I thought it was a calculation on the script side... my bad...but to your question... yes... see attached... That little mini flight used 588Mah of consumption.
[image]

LTMNO, 2018-08-30:
yes thanks.. forgot to close here too ;-)