yaapu/FrskyTelemetryScript #8: Feature Request: No satellites alarm + SatNum sensor

MFlghts, 2018-09-05:
Can you please add voice alarm if sat number drops to 0?
Can you please also add SatNum sensor to be discovered?
Thank you.

yaapu, 2018-09-10:
Hi, I recently removed it, used to be there, I'll add it back :)

yaapu, 2018-09-10:
How about using HDOP instead of satcount wouldn't it provide a better indication of GPS lock quality?

MFlghts, 2018-09-10:
I would like to use it for low sats alarm.
Thank you!

yaapu, 2018-09-10:
What you are looking for is a way to have vocal alert whenever the hdop falls below some threshhold?

yaapu, 2018-09-10:
Ardupilot has a default GPS_HDOP_GOOD of 2.0, I could report low gps hdop whenever it raises above 2 after having been lower

yaapu, 2018-09-10:
having a dedicated menu entry for setting the threshold is IMO overkill and would "eat" some precious memory

MFlghts, 2018-09-12:
My particular case is that sometimes GPS gets jammed in the area I'm flying. I was planing to make an audio alarm using special functions + logical switch to play SatsCount value when it drops to less than 4 sats. Your idea with HDOP is even better! I think any of them will do a job.
Just please make it constantly repeat the warning with 5 sec interval, or so.
Thank you!

yaapu, 2018-09-12:
I'll think about it, ideally there should be a menu entry to enable the feature, then you could configure the periodic alert with the already existing entry. 
Vocal "low HDOP alert" would be nice :-)

btw: do  you get "GPS glitch" alerts on the message bar when GPS gets jammed?

MFlghts, 2018-09-12:
I've just recently started using your brilliant script, I didn't check "GPS glitch" alert yet. Will do next time.
Before I just used OpenTx voice alert for low sats. OpenTx doesn't really natively support all this GPS stuff by default.

MFlghts, 2019-03-24:
Hi yaapu!
I've just installed the last version but seems there is no SatCount sensor in it.
Can you be so kind to add it, as we discussed.

Concerning your question about "GPS glitch" alerts on the message bar when GPS gets jammed - can you please explain what do you mean by GPS gets jammed?
If GPS sats become 0 - voice Alert "GPS lost" is activated. Is there supposed to be some different scenario?

yaapu, 2019-03-25:
Hi,
I did implement the high HDOP alert but was never able to trigger on the field.
I'll try to explain myself, with gps glitches I get "GPS Glitch" but hdop remains high, than "Glitch cleared" and the alert does not fire, if HDOP raises above minimum hdop configured in ardupilot I get an EKF variance error which eventually triggers an ekf failsafe that plays an alert every 10secs.. so I could not find a valid use case for your request.

MFlghts, 2019-03-25:
I want to use SatCount separately to set custom alerts via Taranis voice alarm let's say for 6 sats, 3 sats and 0 sats. This would keep me completely informed about GPS conditions.
Does this sound reasonable?

I might be wrong but somehow I don't like Hdop as it's seems to me that it's easer and more straightforward to control sats amount.

MFlghts, 2019-04-24:
Hi! Can you please inform about any feature for this discussion?

yaapu, 2019-04-24:
Hi,
I will not include sat count sensor in the mainstream code but I can make a special version for you that you'll have to maintain

MFlghts, 2019-04-24:
This is very kind of you! I greatly appreciate it. 
I also had another idea which you didn't reply here ttps://github.com/yaapu/FrskyTelemetryScript/issues/20  , and it's very much connected to this one. Would be very nice if you can also tell will it be also implemented, or denied.
Thank you!

yaapu, 2019-06-24:
Hi,
sorry for a very late replay, install latest version and overwrite yaapu9.luac with this 
[yaapu9_nsats.zip](https://github.com/yaapu/FrskyTelemetryScript/files/3321742/yaapu9_nsats.zip)

Alex

MFlghts, 2019-06-29:
Thank you very much indeed!
I will try asap.

MFlghts, 2019-07-24:
I'm very sorry, can you please compile it to yaapu7?

closed 2021-09-29 by yaapu