yaapu/FrskyTelemetryScript #17: MavLink Passthrough no telemetry data on X9D+

KadaverJoe, 2019-01-03:
I connected my ULRS 1W module to a teensy 3.2 like shown in the ground side picture. If I connect the teensy to the Arduino Serial monitor, I can see Mavlink messages.
I set up the MavlinkToPassthru_v1.0.8 script as follows:
#define Target_Board 0
#define Ground_Mode
#define Battery_mAh_Source 2

These are the messages, which I can see in the serial monitor:
Mavlink in #30 Attitude:  ap_roll degs=-0.8 ap_pitch degs=-0.3 ap_yaw degs=-86.3
Mavlink in #74 VFR_HUD: Airspeed= 1.35  Groundspeed= 0.10  Heading= 273  Throttle %= 0  Baro alt= 36  Climb rate= 0.03
Mavlink in #147 Battery Status:  bat id= 0 bat current mA= 270 ap_current_consumed mAh= 54 my di/dt mAh= 1
Mavlink in #27 Raw_IMU: accX=-0.04  accY=0.01  accZ=-1.00
Mavlink in #29 Scaled_Pressure:   press_abs=1029.4hPa  press_diff=0.000hPa  temperature=35.1C
Mavlink in #30 Attitude:  ap_roll degs=-0.7 ap_pitch degs=-0.3 aMavlink in #30 Attitude:  ap_roll degs=-0.7 ap_pitch degs=-0.2 ap_yaw degs=-86.3
Mavlink in #27 Raw_IMU: accX=-0.04  accY=0.01  accZ=-1.00
Mavlink in #29 Scaled_Pressure:   press_abs=1029.5hPa  press_diff=0.000hPa  temperature=35.1C
Mavlink in #1 Sys_Status:  Bat volts=18.851  Bat amps=0.3  mAh=0.037125  Total mAh=1.188  Bat1 cell count= 5
Mavlink in #125 Power Status: Vcc= 5019  Vservo= 5097  flags= 1
Mavlink in #24 GPS_RAW_INT: ap_fixtype=3 3D Fix  sats visible=15  latitude=54.1732979  longitude=12.3405924  gps alt amsl=25.4  eph (hdop)=77  epv (vdop)=127  vel=0.130  cog=346.6
Mavlink in #65 RC_Channels: Channel count= 16 values:  1=1493 2=1494 3=982 4=1494 5=1494 6=1494 7=1494 8=1817 9=1494 10=1494 11=1494 12=1494 13=1494 14=1494 15=1494 16=982  Receive RSSI=49.21
Mavlink in #74 VFR_HUD: Airspeed= 1.32  Groundspeed= 0.13  Heading= 273  Throttle %= 0  Baro alt= 36  Climb rate= -0.17
Mavlink in #30 Attitude:  ap_roll degs=-0.7 ap_pitch degs=-0.2 ap_yaw degs=-86.3
Mavlink in #147 Battery Status:  bat id= 0 bat current mA= 270 ap_current_consumed mAh= 55 my di/dt mAh= 1
Mavlink in #30 Attitude:  ap_roll degs=-0.7 ap_pitch degs=-0.3 ap_yaw degs=-86.3
Mavlink in #74 VFR_HUD: Airspeed= 1.31  Groundspeed= 0.13  Heading= 273  Throttle %= 0  Baro alt= 36  Climb rate= 0.45
Mavlink in #0 Heartbeat: ap_type=1  ap_autopilot=3  ap_base_mode=89 ap_custom_mode=11  ap_system_status=5  ap_mavlink_version=3
Mavlink in #0 Heartbeat: ap_type=27  ap_autopilot=8  ap_base_mode=4 ap_custom_mode=0  ap_system_status=1  ap_mavlink_version=3
Mavlink in #33 GPS Int:  ap_lat=54.173298 ap_lon=12.340592 ap_amsl=36 ap_alt_ag=-0.0 ap_vx=0.00 ap_vy=0.00 ap_vz=0.00 ap_gps_hdg=273.7
Mavlink in #27 Raw_IMU: accX=-0.05  accY=0.01  accZ=-1.00
Mavlink in #29 Scaled_Pressure:   press_abs=1029.4hPa  press_diff=0.000hPa  temperature=35.1C
Mavlink in #1 Sys_Status:  Bat volts=18.852  Bat amps=0.3  mAh=0.045344  Total mAh=1.233  Bat1 cell count= 5
Mavlink in #125 Power Status: Vcc= 5027  Vservo= 5102  flags= 1
Mavlink in #24 GPS_RAW_INT: ap_fixtype=3 3D Fix  sats visible=15  latitude=54.1732979  longitude=12.3405924  gps alt amsl=25.2  eph (hdop)=77  epv (vdop)=127  vel=0.180  cog=346.6
Mavlink in #65 RC_Channels: Channel count= 16 values:  1=1493 2=1494 3=982 4=1494 5=1494 6=1494 7=1494 8=1817 9=1494 10=1494 11=1494 12=1494 13=1494 14=1494 15=1494 16=982  Receive RSSI=49.21

But if I start my Taranis, the yaapu telemetry page tells me "no telemetry data". Is there anything I did wrong?

yaapu, 2019-01-03:
Hi,
my first wild guess is to delete and rediscover RSSI sensor on OpenTX, in ground mode that sensor is "faked" and is a relay of the LRS link quality.

KadaverJoe, 2019-01-03:
Hi yaapu,
thanks for the info. I deleted the RSSI and tried to rediscover it, but nothing happened.
I only realized that if I change the ULRS to "RX" there are different options. I changed it to RX but then no Mavlink data are seen in the ULRS CC. When I change it to TX, I can see Mavlink packets.

yaapu, 2019-01-03:
do you see any sensor at all on OpenTX? while the teensy is running do you have a valid (high) RSSI?
I display "no telemetry data" when RSSI is 0 or when I do not receive valid frsly passthrough telemetry packets.

Unfortunately I'm not an ULRS user, you'll have to seek for Eric's help on that subject!

closed 2019-01-24 by yaapu