yaapu/FrskyTelemetryScript #18: Sharing sportTelemetryPop() data with other scripts?

KenLagoni, 2019-02-05:
I'm working on another project where I would like to transfer information to and from the receiver via the sportTelemetryPop and Push functions in my own Lua script. However I would also like to have the Yaapu script running and Thus I see a potential problem if two scripts are using the sportTelemetryPop() function, because then WHO is getting the data?

I think the easiest way is to add my decode code in the Yaapu processTelemetry() function and then store the results in a global variable somehow. 
Does anyone know if this is possible? Or is there a simpler way to somehow share or not fight over the telemetry pass-through data?

Thanks.

yaapu, 2019-02-05:
Hi,
I'm afraid the queue is shared so the data belongs to whoever pops it first.
What radio are you targeting?

Memory on Taranis radios is very constrained, there's very little left ro run a second script.
On the Horus you might be able to run an extra widget or two for memory is less constrained.

KenLagoni, 2019-02-05:
I have a Horus where I run a Lua scripts which shows google maps screenshots and plotting the position of the plane.

My script is working because the GPS coordinates are shared like a GPS sensor, and thus data is available in my script. But now I need some more data from the receiver and I’m trying doing it using my own 0x5xxx messages in pass-through, but using numbers which doesn't collide with our script.
Since I don't need speed (only 2-5 times a minute), perhaps just wring data to the SD Card might be a good one to try, I just thought that I could make the data available using a couple of global variables.

Making proxessTelemetry() external was also in my consideration, but it may be overdoing it.

Thanks for the inputs!.

yaapu, 2019-02-05:
Mhmm you could use sportTelemetryPush for uplink and custom frsky sensors downlink?

closed 2019-05-13 by yaapu