yaapu/FrskyTelemetryScript #22: Battery voltage limited to 50V volts

yanivasy, 2019-04-01:
My platform uses an 12 cells baterry reaching 53 volts.
While using the FrSky telemtry script it seems that it can't display more than 50 volts.
As battery discharges bellow 50V telemetry voltage readings seems fine.
Any suggestions?

yaapu, 2019-04-01:
Hi,
I just tried to simulate it and I cannot reproduce your issue, what is your setup?
perhaps dual chained FLVSS or just a PM module?
Please be detailed as to allow me to reproduce it.
Which versions of my script do you use?
Ardupilot version?

yanivasy, 2019-04-01:
Hi Alessandro,
I'm using Pixhawk 2 (which have two independed power inputs).
Script 1.7.2
Mission planner: 1.3.7014.32711
APM Copter V3.5 dev (8c64728f)
PM is based on the original Pixhawk 2 min power brick with additional BEC
to handel the 52V input

Regards,
Yaniv.

yaapu, 2019-04-01:
Hi,
you hit a limit of the telemetry library, if voltage is bigger than 51.1 volt, (511 decivolts expressed as 9 bits) voltage wraps and has to be added to 51.2 to get real voltage.
I'll send you a fix but the script needs to know the cell count and won't be able to autodetect it, so please select 12s in cellcount override in the config menu.

What radio do you have?

yanivasy, 2019-04-01:
Hi,
I'm using Taranis X9D plus.
I manually entered 12s I the script configuration menu before publishing he
issue but with no change.

Thanks,

Yaniv.

yaapu, 2019-04-01:
Hi,
you need to wait for my fix, I have it working on my test setup.

Dual battery detection relies on ardupilot sending second battery info.
The script relies on the parameter [batt2_monitor](http://ardupilot.org/copter/docs/parameters.html#batt2-monitor-battery-monitoring)

yanivasy, 2019-04-02:
All bat2_monitor  parameters are set within Pixhawk 2 so Mission Planner displays can both batteries.
What I can't figure out is how to display the second battery input using your srcipt on Taranis.

yaapu, 2019-04-02:
ok, short press [enter] to enable dual battery view

yanivasy, 2019-04-02:
When I shortly pressing [enter], up and down arrow apears  showing higher or lower values that seems like min/max values.
What am I doing wrong?

yaapu, 2019-04-02:
Hi [enter] is bottom-right, if you see up or down arrow you're pressing [menu] top-left

[image]

yaapu, 2019-04-02:
Hi,
here's the [fix](https://drive.google.com/open?id=1KlvJx8_t-aZ_lQSUFE4x_ci5O8984DSp), please copy over existing files.
Let me know how it goes :-)

Alex

yanivasy, 2019-04-04:
Hi Alex,
After changing manually the cell nuber to 12 (as you noticed), your fix is working good - Thanks.
As for the second battery display, pressing [enter] togells between the following:
4.25V FC 51.0V (bat 1)
0.00V a2 0.0V
0.00V vs 0.0V
I can't seen the second bat monitor altought it is shown correctly on Mission Planner
What am I doing wrong?

yaapu, 2019-04-04:
mhmmm, ok, 
looks like your setup sends 2nd battery info in a way I do not detect, could you please run a special version of my script that dumps all the telemetry to a file.
What I need: 
- a tlog from MP to actually see the 2nd battery is detected
- a plog file produced by this [debug](https://drive.google.com/open?id=1k83JvV8T_FbeI5xwufv_i7C8Yu-yWxyu) version of my script, the log file is saved in /LOGS/<model + date>.plog

yanivasy, 2019-04-07:
Hi Alex,
see the attached files

yaapu, 2019-04-07:
Hi,
I don't see any attachment!

yanivasy, 2019-04-07:
[2nd batt.zip](https://github.com/yaapu/FrskyTelemetryScript/files/3051898/2nd.batt.zip)
You are right, couldn't uploaded them by Explorer.
Now I'm using Chrome

yaapu, 2019-04-08:
Your flight controller is not sending 2nd battery info at all, I checked ardupilot code and support for 2nd battery was introduced in Copter 3.6, this explains it all, sorry, looks like you're only way to solve this is switch to Copter 3.6

yanivasy, 2019-04-10:
Hi Alex,
I've upgraded to copter 3.6.7 and now I can see in  B2 zoon (left side of Taranis's screen) only sum of both baterries (B1+B2). Since in my case Bat1=51V and bat2=18V, he sum is 69V.
Beside that, the capaticy of both baterries displays the same 6.5Ah altought in MP different capaticies are shown (13.1Ah & 3.3Ah).
Once again, on MP everthing is display correctly so what am I doing wrong?

yaapu, 2019-04-10:
Hi,
I don't think you're doing anything wrong, could you please send me a plog from your new setup so I can reproduce your issue, please use the [debug](https://drive.google.com/open?id=1k83JvV8T_FbeI5xwufv_i7C8Yu-yWxyu) version to save it.
I'm sure we can get this fixed!

yanivasy, 2019-04-10:
As you suggested, I used the debug version and now both batteries are displayed correctly (since Bat1 is currntly lower than 52V).
This implies that the 52V fix has somthing to do with it.
I'll send debug files later since I'll need a chrome browser...

yaapu, 2019-04-10:
It should not have any impact but I'll check anyway. Can you also describe your battery setup in detail as per my previous post? It would allow me to better understand the issue.

yanivasy, 2019-04-10:
[log2.zip](https://github.com/yaapu/FrskyTelemetryScript/files/3065192/log2.zip)
I've attahced a plog file.

The 1st battery input of Pixhawk2 (the cube) is connected to a mini power brick whereas the 5V is produced by a 52V->5V BEC (a DC/DC converter). The mini power brick is only used for voltage and current monitor. To enable higher voltage monitoring, a resistor within the mini power brick was repalced (the mini power brick uses a two resistors voltage divider). The baterry in use is 12s liPo.

As of the 2nd battery input of the cube, it uses a standard mini power brick (without any changes) tht is connected to another DC/DC converter that reduces a 12s LiPo to 18V and thereby can be used by the standard mini power brick. Current is not monitor in this case - only voltage. This is only for backup power to the cube.

yaapu, 2019-04-12:
Hi,
well you have 2 independent batteries :-)
This [version 1.8.0 beta5](https://drive.google.com/open?id=140mFnxN7_bLSF7k73CuwgjZOTHlm_Pkb) adds support for your battery setup.

Go into the config menu and set:
 dual battery setup: other
 batt[1] cells override to 12s. 
 batt[2] cells override to 1s. (show voltage as 18V)

this version also has a lot of new features, you can check them by looking at this [thread](https://discuss.ardupilot.org/t/an-open-source-frsky-telemetry-script-for-the-horus-x10-x12-and-taranis-x9d-x9e-and-qx7-radios/26443/587) on discuss

yaapu, 2019-04-12:
I fixed an issue in the battery % vocal alerts, please use [this one](https://drive.google.com/open?id=140mFnxN7_bLSF7k73CuwgjZOTHlm_Pkb)

yanivasy, 2019-04-16:
Hi Alex,
It seems that file copter.lua was missing in scripts/yapuu library.
After using the old one its working well.
Maybe for selecting the extra screen, if possible, it will be easier to press the minus button.
Can't find a table describing all sensors name available to the extra screen.
Thanks.

yaapu, 2019-04-16:
Hi,
I checked the zip file and the required copter.luac file was there.
What was the error you had that made you copy the copter.lua file over?

> Maybe for selecting the extra screen, if possible, it will be easier to press the minus button.
Yes I know, I'm looking for a better way to switch to the extra screen too

As for the sensors, you can add any sensor that you already have as a standard frsky sensor in your telemetry page, for example if you had the FLVSS cell monitor you could add the 
"Celd" to monitor max cell delta voltage

yanivasy, 2019-04-16:
copter.luac file exist in the zip file but copter.lua doesn't.
The error says simple that this file (copter with lua extension) is missing so I just added it.
I want to add the two voltage monitors (bat1 & bat2), current monitor, sat number, altitudes from both GPS and ground.
BTW, is it possible to see the second GPS telemetry also?

yaapu, 2019-04-16:
> I want to add the two voltage monitors (bat1 & bat2), current monitor, sat number, altitudes from both GPS and ground.
BTW, is it possible to see the second GPS telemetry also?

Hi,
none of this is possible sorry, only proper frsky sensors.

yanivasy, 2019-04-17:
This applies to the the 2nd GPS telemetry also?

yaapu, 2019-04-17:
yes, the ardupilot telemetry library does not send any info about 2nd GPS

closed 2019-05-13 by yaapu