yaapu/FrskyTelemetryScript #23: Support for Betaflight

backman8571, 2019-04-23:
I have not seen if this will work with Betaflight

Also nothing here on setting it up for the Jumper T16 radio, is it the same as the 10S ??

Could you create a video for this

Per

yaapu, 2019-04-23:
Hi,
this is for ardupilot only, and the X10 version "should" work on T16

you can follow the developemnt [here](https://www.rcgroups.com/forums/showthread.php?3020527-An-Ardupilot-frsky-telemetry-LUA-script-for-the-Horus-X10-X12-and-Taranis-X9D-E-QX7)

as for the video, well I'm waiting for someone to create a nice one :-)

backman8571, 2019-04-23:
Alessandro, is it possible for you to make a version for BetaFlight.

I think its more users for betaFlight then Ardupilot

And great job by the way

Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

LTMNO, 2019-04-23:
I would beg to differ on that topic.  ;-)

backman8571, 2019-04-23:
Here in Texas at my club and shop's over 100 and 2 using  Ardupilot
Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

backman8571, 2019-04-23:
How much work would it be to make it compatible with BetaFlight??

Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

backman8571, 2019-04-23:
Also i have the Tyranis 9D and 7S

Do you have a procedure of how to get the scripts running on them 2 ??

Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

yaapu, 2019-04-26:
Hi,
sorry but I won't make a betaflight version, you might checkt the iNav lua scripts and ask over there for betaflight compatibility!

You can find docs on how to install my scripts in my [readme](https://github.com/yaapu/FrskyTelemetryScript/blob/master/README.md#installation-on-taranis)

closed 2019-04-26 by yaapu

backman8571, 2019-04-26:
I have no clue of how to do that

Regards

Per Backman

Systems Management/Network Management
Professional Services

CELL : 4698558729

Email1 : per.backman@aurea.com
Email2 : per.backman@zephyrtel.com
SKYPE: backman8571@live.com

yaapu, 2019-04-26:
Hi, 
go [here](https://github.com/iNavFlight/LuaTelemetry) and ask for support.
Betaflight and iNav are much more similar than betaflight and Ardupilot.

Alex.