yaapu/FrskyTelemetryScript #26: FrSKY X9D PLUS Taranis "not enough memory"

Kadam23, 2019-06-18:
Dear,
i installe script on my FrSKY X9D PLUS Taranis. Starting script report "not enough memory"

yaapu, 2019-06-18:
Hi,
this happens when you try to compile the script on the radio.
You need to copy the binary version from https://github.com/yaapu/FrskyTelemetryScript/tree/master/TARANIS/SD 
to your radio SD card.
Then configure the script as a telemetry one. 
Make sure it's the only script running.

yaapu, 2019-08-08:
is this solved?

Kadam23, 2019-08-09:
Dear,
I've to try removing other plugin. I let you know.
Thx

closed 2019-10-30 by yaapu

StrategicVisionary, 2023-03-06:
I would confirm and reemphasize make sure the Yaapu script is the only one running. My memory error problem abated after removing the farl script on the second telemetry page.

yaapu, 2023-03-06:
...another thing to watch for is OpenTX vs EdgeTX with the latter (version 2.8.x) leaving much less memory to LUA scripts

srhassan71, 2023-12-26:
@Alex, does this mean EdgeTX + ELRS on Taranis X9D (ACCST) cannot run Yaapu?
I don't have any other scripts running, and when I try to enable CRSF protocol (Yes/No), it gives me an error "Not enough memory"

ZSeymour1, 2025-07-04:
I've had this same problem. My mistake was to copy the contents of the SRC folder into the SD card's SCRIPTS/TELEMETRY folder - don't do that! Hope this helps :)