yaapu/FrskyTelemetryScript #27: Expose Flight Mode numeric value from ArduPilot MavLink to OpenTX

JoshWelsh, 2019-06-24:
Hey there.. we have a need to do a mix in the radio that is dependent on the ArduPlane Flight Mode numeric value.  Is it possible to expose this to OpenTX from your widget as a sensor along with the others that you're exposing?

yaapu, 2019-06-24:
Hi Josh,
sure it's an easy mod you can do yourself, what radio do you have?

JoshWelsh, 2019-06-24:
Hey Alessandro, thanks for asking, running a Horus.

yaapu, 2019-06-24:
Hi,
add this line 

`setTelemetryValue(0x060F, 0, 2, telemetry.flightMode, 0 , 0 , "FMOD")`

as last instruction of this function
https://github.com/yaapu/FrskyTelemetryScript/blob/master/HORUS/SOURCES/PP/main.lua#L1173

and copy main.lua to the WIDGETS folder

cheers,

Alex.

JoshWelsh, 2019-06-24:
Hi Alessandro,

I’ve done what you instructed here:

https://github.com/JoshWelsh/FrskyTelemetryScript/blob/1102effc825ebf4492e80fcf23b33ca55d7d8e81/HORUS/SOURCES/PP/main.lua#L1208

When I copied the file over, it reverted the widget back to 1.7.4 from 1.8rc2, whos main.lua appears to be binary?  I think I may have just messed the whole thing up though, can you advise?

https://github.com/yaapu/FrskyTelemetryScript/blob/master/HORUS/SOURCES/PP/main.lua#L1173<https://nam03.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fyaapu%2FFrskyTelemetryScript%2Fblob%2Fmaster%2FHORUS%2FSOURCES%2FPP%2Fmain.lua%23L1173&data=02%7C01%7Cjosh%40joshco.us%7C28127854d9a246c02a4e08d6f8c00992%7C66a80cae9a51453283356eed78d28e07%7C1%7C0%7C636969901020653158&sdata=d0fuSH3HGiOKkMQ%2FuRxVH952VYqNVnEInwTKkZJPr5E%3D&reserved=0>

yaapu, 2019-06-25:
mhmm my fault, you have told you it was based on 1.7.x, try this patched version anc copy it to the widgets/yaapy/folder

[yaapu_widget.zip](https://github.com/yaapu/FrskyTelemetryScript/files/3325007/yaapu_widget.zip)

JoshWelsh, 2019-06-25:
My apologies sir, I should have been more clear. Thank you for this, will try it later today and report back.

yaapu, 2019-10-30:
Josh I'm closing this, if you need more help open a new issue!

closed 2019-10-30 by yaapu