yaapu/FrskyTelemetryScript #28: Taranis-1.8.0: slowness on 9xd+ with opentx 2.2.3

zingaros, 2019-06-26:
Ciao,

First fantastic work: having all this telemetry with no additional hardware except an omnibus pro F4 and the R9mm is just a revolution in the ardupilot world.

Now I did some test with the previous release and it was working flawlessly then I decided to test the last stable release (1.8) My issue is that the voice is very laggy: almost a second between the change of a flight mode and it’s announcement. On screen it’s in time (instant) but the voice lags. 

Should I do something particular ? I enabled Lua and Luac (looks luac is a new requirement) I also have only this sole script running.

Many thanks

yaapu, 2019-06-27:
Hi,
thanks for the kinf words!
It must be something specific to your setup, I have a pretty close configuration with ominibusf4 + r9mini + Taranis X9D on OpenTX 2.2.3 plane 3.9.8/copter 3.6.7, yaapu script version 1.8.0 so I know for sure it has to work!
You're the first one to report audio lag, try to describe you're setup better.
OpenTX and Ardupilot version, detailed wiring, max3232 cable?
Do you experience the lag only after arming?

zingaros, 2019-06-27:
Hello,
Here is a video https://youtu.be/VctXEW2ISpY
This will give you an objective experience: much better than yesterday: I  guess yesterday I have not properly  have shut down the radio and  instead of reboot it / discover sensors.

yaapu, 2019-06-27:
Yes, cannot get much faster than this, in version 1.8.0 there really is a lot more happening in background so audio is a bit slower

closed 2019-07-01 by yaapu