yaapu/FrskyTelemetryScript #29: Transmitter Voltage Warnung.

Basti77, 2019-06-27:
Hello,
the Xlite (the smallest display) lacks the transmitter's voltage or is too inconspicuous.
Is there a way to acoustically maintain the TX battery voltage?

If the LUA script is running, the transmitter does not warn against undervoltage. 
(There is only one exclamation mark)
I ran out of transmitter during flight (R9 module  is very hungry for electricity) 

A Wing just landing over notebook and 433Mhz telemetry is an adventure... but it works.

yaapu, 2019-06-27:
Hi, doesn't OpenTX warn you with low voltage?
I do not monitor tx voltage because I use the built in warning in OpenTX

Basti77, 2019-06-28:
Last year at indoor flying the TX Acoustic has announced itself. 
OTX 2.2 that was just a beep. 
there was no LUA script running. you say immediately show the battery and Ohh stop action. 
Now I'm at 2.23 because of R9.
I can't tell now if the FW doesn't give a beep anymore (without LUA Script) or if general scripts with MP3 support block the native sound output of the TX.  

I need to check the battery for any damage caused deep discharge. 
I don't want to make the batteries empty again.

yaapu, 2019-06-28:
Hi,
I'm sure that I get low battery warning from my Horus and my Taranis, just asking if you do get low battery warning from your xlite, I'm sorry for your battery :-(
to recap:
- opentx 2.2.2 you remember hearing opentx low battery beeps
- opentx 2.2.3 you did not hear low battery opentx warning, it should be a sound in wav format

if this is correct it's a serious issue in OpenTX, while lua is running all system sounds must be active.

If you have a variable power supply you can try to simulate a battery running low by attaching it to the xlite battery plug and gradually lowering the battery until you trigger low battery warning...

Basti77, 2019-06-28:
I'll keep investigating. the problem is not unimportant. ;)
But still the voltage from the TX battery is missing in the small display.
please compare the screenshots.

yaapu, 2019-06-28:
yes I know, I removed the rx voltage in version 1.8, I'll look at a way to put it back in perhaps as small bars

yaapu, 2021-04-18:
fixed in latest 1.9.x

closed 2021-04-18 by yaapu