yaapu/FrskyTelemetryScript #30: Hope to use this LUA on FLYSKY Nirvana14

dengboheng, 2019-07-08:
Nirvana 14 has the same level of hardware configuration as frsky Hours X10 and also runs Opentx.

yaapu, 2019-07-08:
Hi,
well kind of hard :-)
Different lcd aspect ratio, and it's not officially supported by OpenTX nor by OpenTX Companion.
The OpenTX nirvana build is not even public!

yaapu, 2019-08-08:
I'll leave this open waiting for more info from the OpenTX team

yaapu, 2019-09-26:
@dengboheng what flight stack do you run? Ardupilot?

yaapu, 2019-09-26:
@qba667 is the DIY multi module supported on the NV14?

qba667, 2019-09-29:
@yaapu I added support for multi-module last Friday.
It will be officially merged into 2.4 - but it is hell of work...

yaapu, 2019-09-29:
@qba667 that is good news and I must say you're doing a great job!
I know your work on the FS-I6 and on flysky telemetry so I'm pretty sure you're up to the task :-)
Can you give a pointer to your repo so I can build the latest version?

My widget will crash a real radio for sure because of off screen writes but the simulator can probably survive such calls :-)

qba667, 2019-09-30:
@yaapu  try this out:
https://github.com/open-flysky/Flysky-OpenTX-Test-Branch
I will also create a test build with this feature, but first I need to double check if all external modules are working. Just let me know in case of any questions.

yaapu, 2019-09-30:
@qba667 thanks, target simu is working fine, once you release multi module support I'll need a real radio to do proper testing, simulators are fine but a bit forgiving, especially on the memory side :-)

qba667, 2019-09-30:
@yaapu multi is out - at lest test build. I will ask FlySky about radio for you, but probably in next week because now they have public holidays for more than 1 week. 
Regards from somewhere north from you:)

yaapu, 2019-09-30:
> I will ask FlySky about radio for you, but probably in next week because now they have public holidays for more than 1 week.

@qba667 that would be absolutely great, thank you.

It's not clear to me how your OpenTX version relates to other I see around, yours is 2.1.8 while I see latest UndergroundFPV is 2.2.2, mine it's just curiosity since I haven't been following NV14 threads.

more thanks from somewhere south of you ;)

qba667, 2019-10-01:
It is all based on 2.2.2 because of that UFPV is named 2.2.2, FlySky decreased the version number from some unknown reasons (maybe feature set not complete). What is in this repository is something between 2.2 and new official 2.4 because 2.4 will be based on new UI that was introduced in this project.

yaapu, 2019-10-01:
This totally makes sense and explains why the LUA API is aligned with OpenTX 2.2.x feature set

qba667, 2019-10-10:
@yaapu touch events are now passed to widgets.
https://github.com/open-flysky/Flysky-OpenTX-Test-Branch/tree/widgets_events
This should give us other dimension of usability.

yaapu, 2019-10-10:
@qba667 congrats, this is sweet!

I glanced through the code, your passing events to the selected widget only and if I touch another widget you select it first, correct?

Did you define gestures as well or plain touch events with coordinates and perhaps touch "duration"?

qba667, 2019-10-15:
@yaapu that is right. Over weekend I tried to force the touch controller to produce gestures, unfortunately it won't output anything. There is already existing TouchManager class that can be use to detect and translate touches into gestures. This must be addressed anyhow.

yaapu, 2019-10-15:
@qba667 gestures will be needed when the lcd subsystem will be fast enough to zoom and rotate in real time which I'm afraid will be a long time from now :-) right now it's hard to keep a 10Hz fps

Bas-moz, 2020-09-11:
@yaapu Is there any status updates on this, I have an NV14 and I'd be interested in contributing the code needed for this but I'm having some trouble figuring out the specifics of the build system, for one is it possible to build and run this without using the companion app compiler? As the companion app doesn't support the NV14 as you've mentioned earlier. Mind you, I've never owned another OpenTX RC so I'm not even sure what the process is supposed to look like :-).

Bas-moz, 2020-09-11:
@qba667 I compiled the https://github.com/open-flysky/Flysky-OpenTX-Test-Branch/, and I got it running on my NV14 just fine. However widgets are still a mystery to be, the included Counter and BattCheck widgets don't work. They complain about refresh() trying to call a table, this turned out to be about the table returned from create being called. I made that callable but that didn't seem to make them work, as a matter of fact it doesn't seem the refresh lua function is called at all? I'm sure I'm missing something obvious, but it makes it tricky even starting to try and get this working :).

Bas-moz, 2020-09-11:
Alright, appears I've located the bug inside the test branch. I've got a fixed build where widgets work again as expected (bug was introduced in https://github.com/open-flysky/Flysky-OpenTX-Test-Branch/commit/624101cf5d1f21ea1275c2a4828b15e8c02e89e1). I'll attempt to open an issue over there. Sorry for polluting this bug with it

yaapu, 2020-09-11:
@Bas-moz Hi, I did not work on this, the simulator was working though so no need for the whole companion app.
Porting to another layout is not an easy task, not impossible but requires a good understanding f the widget

Bas-moz, 2020-09-11:
@yaapu Yeah. I figured :-). I actually got things working and building after I fixed a bug in the firmware code. I have the main HUD and general mesaages working although there's indeed a lot of layout work still to do. Although the HORUS implementation seems to be a little hacky as well I'd like to eventually get it into a state of sufficient quality that it can be upstreamed. Once I get something sort of working I'll post the WIP at least so other NV14 users can play with it. Hopefully I can get the firmware fix in as well. Thanks for your reply!

yaapu, 2020-09-12:
> Although the HORUS implementation seems to be a little hacky as well I'd like to eventually get it into a state of sufficient quality that it can be upstreamed
@Bas-moz  can you elaborate on this, I'm always open to improve my code

Bas-moz, 2020-09-12:
@yaapu By all means! FWIW I've mostly been working on the HUD so far. But I was referring mostly to the strange interaction between the 'widget' (which has its own dimensions), and then overwriting the whole LCD, for the NV14 this seems like it's probably the wrong approach and I would actually constrain the size of the widget to the 'actual' widget on the device. (otherwise you end up overwriting the device's own immutable buttons)

For one file I was also making the mistake of editing the 'preprocessed' file, which made things appear hackier than they actually were once I figured out the directory structure and how to use the preprocessor correctly :-).

markandkymward, 2021-11-25:
Curious if there has been any movement on this.  Just installed EdgeTX on my (now defunct) UndergroundFPV Nirvana.  I always loved this radio but put it aside when things went sideways.  I even moved over a 6pos switch to replace one the the VRB pots.  It is looking good but sadly, no @yaapu YAAPU.  I did manage to get your telemetry script working, albeit wonky in appearance.

yaapu, 2021-11-26:
Hi, indeed now it should be doable, I don't have a radio to test it though. I guess I will eventually support the NV14 on edgetx

markandkymward, 2021-11-27:
Aside from all the drama that came with it, it really was a great radio. Awesome to see it get a second life in EdgeTX.   If I can assist, I'd be happy to help.

yaapu, 2022-12-31:
finally getting there

[image]

lin554893462, 2023-01-24:
EL18有可用的遥测脚本吗

lin554893462, 2023-01-24:
Is there a telemetry script available for EL18?

yaapu, 2023-01-24:
> Is there a telemetry script available for EL18?

not ready yet, very soon

lin554893462, 2023-01-25:
Wait for your good news

yaapu, 2023-02-01:
hi all, please download latest zip from github and test :-)

closed 2023-02-01 by yaapu

lin554893462, 2023-02-01:
ok

 
Closed #30 as completed.

lin554893462, 2023-02-01:
okay, thank you