yaapu/FrskyTelemetryScript #32: QX7 OpenTx 2.2.4 memory issue

LubinKerhuel, 2019-08-05:
Syntax error (attempt to call a nil value) when QX7 has the OpenTx 2.2.4 with lua and luac options enabled.

Not enough memory (or panic) when QX7 has the OpenTx 2.2.4 with only lua firmware. 

I might have switch too early on openTx2.2.4 version ;-)
Nice work !

yaapu, 2019-08-05:
I tested it on 2.2.4 on my X9D.
Your radio is probably trying to compile the script.

You need already compiled lua files, use only the binary files here, lua or luac:
https://github.com/yaapu/FrskyTelemetryScript/tree/master/TARANIS/SD/SCRIPTS/TELEMETRY

If you had a working 2.2.3 setup and upon switching to 2.2.4 you got an error please try with a new OpenTX model from scratch

LubinKerhuel, 2019-08-05:
thanks for quick response.
I did not tested with 2.2.3 unfortunately.

Which firmware option is best to use ? (did you enabled luac ? use the alternate font etc ?)
I think the X9D is based on the same MCU as the qx7 thus I should be able to run the script.

yaapu, 2019-08-05:
Hi,
enable luac and if you like you can use custom fonts.
As you said your QX7 should run the script just fine!

LubinKerhuel, 2019-08-06:
ok, I get it working after removing the lua option, keeping only luac and noheli.

yaapu, 2019-08-06:
Ok great!
Glad it works :)

closed 2019-08-06 by yaapu