yaapu/FrskyTelemetryScript #33: Show last good GPS coordinate on Taranis

zingaros, 2019-08-06:
Hello

Yesterday I lost my plane in a corn field 400 m away from me. Lost telemetry. I was running Yaapu: I would have so much appreciated to have the last coordinates shown instead of a lot of data. I had to spend a lot of time to play pause the recorded video and found it but... but...
Thanks !

yaapu, 2019-08-06:
Hi, I'm really  sorry for your crash.

GPS coordinates on Taranis radios can be shown easily (and logged to the SD card just as easy) by using built in OpenTX functionality, no need to have the yaapu script handle them.

This is standard OpenTX stuff but to help set it up I wrote a [guide](https://github.com/yaapu/FrskyTelemetryScript/wiki/How-to-display-GPS-coordinates-on-Taranis-radios) in my wiki, check it out and let me know what you think!

zingaros, 2019-08-07:
Thanks Alessandro

Your wiki page is perfect, especially the trick with the DMS format, but imo we have a but... 
if I loose telemetry during the crash while using yaapu on a screen, when I crash, I switch to the other  gps screen that will say “no telemetry”. 
It means I will have to take a laptop to read the logs on the SD card: not very convenient.
Do you concur ?

yaapu, 2019-08-07:
No, sensor values are retained even after losing telemetry. After a link loss they are simply not updated anymore and showing last values

zingaros, 2019-08-07:
Thanks and sorry to had doubts !

closed 2019-08-08 by yaapu