yaapu/FrskyTelemetryScript #34: Feature request: Mavlink telemetry data on Jumper T16 (Ultra LRS system)

Molecuman, 2019-08-08:
Hello

Please make the widget compatible to OrangeRX Ultra long range system with a Pixhawk sending mavlink data thru serial port. In other words, pure mavlink with no FrSky in between. 
For more info: http://www.itluxembourg.lu/site/orangerx-modules-pinout/using-the-new-hk-orangerx-modules-with-bluetooth/

Thank you

yaapu, 2019-08-08:
Hi,
please take a look at this [project](https://github.com/zs6buj/MavlinkToPassthru)

Molecuman, 2019-08-08:
Hello. Thank you! I have a lot to read ;)

Molecuman, 2019-08-08:
Hello

Sr. I´m sorry but I really don´t understand what to do. I´m very nubbie on
this and I just don´t understand what to do with those files. If you could
explain me please: Do I need more hardware to maje the telemetry work with
the Jumper T16? Or can be done in a more easy way?

Sorry, I have read a lot and I don´t know what to do

Consider that my native language is spanish.

If you can help me I´ll be glad. Thank you!

yaapu, 2019-08-09:
you need to get one of the supported boards

https://github.com/zs6buj/MavlinkToPassthru/wiki#suitable-ucontroller-boards

configure and flash it with the custom firmware.
I suggest you ask for help in the proper thread on rcgroups

https://www.rcgroups.com/forums/showthread.php?3089648-Mavlink-To-FrSky-Passthrough-Converter

Molecuman, 2019-08-09:
Ok, thank you again Sr.

closed 2019-08-11 by yaapu