yaapu/FrskyTelemetryScript #40: AltAsl parameter

kpihus, 2019-10-25:
Hi,

https://raw.githubusercontent.com/yaapu/FrskyTelemetryScript/master/HORUS/IMAGES/x10displayinfo.png
According to this image, it seems that AltAsl should display absolute altitude above sea level when rangefinder max is not set (this rangefinder mode not enabled) and if rangefinder max is set, it should display rangefinder measurement. 

I have rangefinder max enabled but i can still see gps altitude there. QGC reports rangefinder measurement correctly.

my setup:
px4@1.9.0
Jumper T16 with zs6buj/MavlinkToPassthru converter

yaapu, 2019-11-11:
Hi, I finally had time to look at this and you're absolutely right, I must have broken this some how.
I'll push a fix ASAP.

thank you!

yaapu, 2019-11-11:
Hi could you please try if this fixes the issue?

[horus_1.8.1_rc1.zip](http://bit.ly/34QfSLm)

closed 2020-02-22 by yaapu