yaapu/FrskyTelemetryScript #41: S.Port Telemetry

mric3412, 2019-11-02:
This issue is a new feature propsition.9
Would it be possible to manage S.Port Telemetry? Then, it may work without any converter with Lora-LRS for example (http://qczek.beyondrc.com/qczek-lrs-433mhz-1w-lora-rc-link/qczek-lrs-sport-telemetry/) It seems to be implemented in an opensource Lua Telemetry script for Inav (https://github.com/iNavFlight/LuaTelemetry) Would it be possible to merge this S.Port functionality for ardupilot ?

mric3412, 2019-11-04:
Browsing the code, I realized the complexity of this great work (and I never used Lua...)
If I understand well, it should work well with ARDUPLANE(mavlink) connected to Lora-QczekLRS-Rx(mavlink) and Lora-QczekLRS-Tx(mavlink) connected to ESP32 (mav2passthru in ground mode) plugged in JumperT16/Taranis. Is it correct ?
Since I tested a simpler configuration working for Inav (Inav Lua telemetry deals directly with S.Port encapsulation of QczekLRS-Tx without any converter), I'm just wondering if it would be feasible to adapt processTelemetry() in order to manage directly these datas send by QczekLRS S.Port encapsulation : 
APID_GPS_COURSE 0x0840
APID_RSSI 0xF101
APID_VFAS 0x0210
APID_CURRENT 0x0200
APID_CELLS 0x0300
APID_ALTITUDE 0x0100
APID_VARIO 0x0110
APID_GPS_SPPED 0x0830
APID_LATLONG 0x0800
APID_GPS_ALTITUDE 0x0820
APID_AIR_SPEED 0x0a00
APID_FUEL 0x0600
APID_T1 0x0400
APID_T2 0x0410
APID_PITCH 0x0430
APID_ROLL 0x0440
APID_MAV_BASE_MODE 0x04A0
APID_MAV_SYS_STATUS 0x04A1
APID_MAV_CUSTOM_MODE 0x04A2
APID_CUST_RSSI 0x04B0
APID_RX_RSSI_REG_VAL 0x04B1 – with offset 157 for 868MHz and 146 for 433MHz
APID_RX_SNR_REG_VAL 0x04B2 – with offset 64.
APID_RX_PACKET_LOST_VAL 0x04B3
Any advice (or link to any documentation) to better understand this system would be welcome ; 
Thanks for this nice work !

yaapu, 2019-11-11:
Hi,
sorry for the delay.

> If I understand well, it should work well with ARDUPLANE(mavlink) connected to Lora-QczekLRS-Rx(mavlink) and Lora-QczekLRS-Tx(mavlink) connected to ESP32 (mav2passthru in ground mode) plugged in JumperT16/Taranis. Is it correct ?

yes it works fine and that's the recommended setup for long range systems like ulrs,QczekLRS, dragonling and crossfire.

> Any advice (or link to any documentation) to better understand this system would be welcome ;
> Thanks for this nice work !

You need to study the code : -) rewriting processTelemetry() is a possible solution but you would loose quite a bit of features, specifically you loose mavling messages, flight modes and rangefinder support, you also loose waypoints and airspeed support provided by the MavToPT code

mric3412, 2019-11-11:
Thanks for your answer, it's clearer now. It would be a shame to remove features to this nice script! I hope I will have the time to write a "light" processTelemetry() to manage the few datas from QczekLRS/SPort in order to learn Lua and to have a simple telemetry donwlink.

closed 2019-11-11 by mric3412