yaapu/FrskyTelemetryScript #42: No data sent from MatekF405-Wing

dposea, 2019-11-07:
Alex,
I don't think this a bug, but I can not get passthru to work. I have used an FDTI adapter with the RX line tied to the Serial 4 port TX line of the FC, and of course set SERIAL4_PROTOCOL=10 and OPTIONS=7. All I see from the sniff are the receivers requests for data:

dbf 717e ef7e bdbf 537e 347e ef7e bdbf
0002640 e27e 9690 ef7e bdbf ef7e bdbf 397e ef7e
0002650 bdbf e27e 9b90 e27e 8090 ef7e bdbf 227e
0002660 ef7e bdbf ef7e bdbf 457e ef7e bdbf 677e
0002670 487e ef7e bdbf 6a7e ef7e bdbf ef7e bdbf
0002680 0d7e ef7e bdbf 2f7e ef7e bdbf 717e ef7e
0002690 bdbf 537e 347e ef7e bdbf e27e 9690 ef7e
00026a0 bdbf ef7e bdbf 397e ef7e bdbf e27e 9b90
00026b0 e27e 8090 ef7e bdbf 227e ef7e bdbf ef7e
00026c0 bdbf 457e ef7e bdbf 677e 

I added debug prints to the passthru code in ardupilot, in the send_byte method, and I get data. I did not capture it but I try to figure out how to do that, to capture all bytes sent. Nothing shows up on the TX line for FC sends. No converter, just the SERIAL_OPTIONS=7 and a wire soldered to the P pad on an R-XSR receiver. 

Can you offer any insight? It shoudl just work. I've tried the latest dev with your new scheduler, and 3.9.

yaapu, 2020-02-22:
I'm really sorry but for some reason I ovelooked your issue, please excuse me.
Did you ever solve it?

dposea, 2020-02-26:
Thx for the reply anyway :)  I was able to finally get passthru to work using a circuit I found on the internet that used a couple of transistors. I posted the fixed info in several places, the main one being a thread on the ardupilot forums. Clone MAX232 chips just would not work at all. If you want I can find the schematic for the circuit I used. Or just close this and anyone that wants the info can ask me.

yaapu, 2020-02-26:
Hi David,
I had the same issue with fake max3232, I eventually had to swap them all with TI genuine ICs and all issue magically disappeared,

thanks for your feedback

closed 2020-02-26 by yaapu