yaapu/FrskyTelemetryScript #44: Battery persentage always shows 99%

spencer-edh, 2019-11-19:
I have setup this scrip successfully on two pixhawks using my Jumper T16 running version 1.8.0. now and love it. However, my battery percentage is always at 99%. Is there any reason for this? Am I missing a setting somewhere to configure this?

yaapu, 2019-11-20:
Hi,
do you have a battery monitor that can measure current as well as voltage?

spencer-edh, 2019-12-03:
So after flying my drone around looks like i do see the battery percentage drop. When i set the script up i plugged in a battery knowing it was at 3.8V per cell and that was reflected correctly however the percentage showed 99% when I was expecting like 40%. It would be cool to have that percentage bar configureable. So when you plug a battery in and its not full it will display as such. Not sure if its possible to change that functionality or if thats not how you have designed it to work. Just some thoughts.

yaapu, 2019-12-03:
I'm working on a feature that displays battery % by battery voltage, but once enabled voltage would be the only source for %.
You would either use voltage or current but not both because there's no way to know when the pilot plugged the battery in

spencer-edh, 2019-12-05:
sounds good. I have been making some youtube videos on how to set this up and will make more in the future. love the script good work! Here is the video if your interesting in seeing it. https://youtu.be/wAWdZoBrsug

closed 2019-12-05 by spencer-edh

yaapu, 2019-12-08:
Hi,
I saw your video, really nicely done!
If you plan on doing one for the mapping feature I could link it in the wiki.
In the one you published you shoudl probably enable the battery monitor so voltage would not read 0 all the time :-)

spencer-edh, 2019-12-09:
Hey thanks! I absolutely plan on making a video when 1.8.1 has a stable release. Im actually really excited about that feature. Good thought on linking to the wiki! I will for sure include that on my next video. 

haha yes I do have the battery monitor enabled now and will show all that off in the next video I do. 👍

yaapu, 2019-12-15:
I've been flying the mapping feature for a while and so did many I asked to beta test it.
The beta phase is almost done, I haven't received any issue reports on the mapping feature, so I guess you could very well do a video now.

spencer-edh, 2019-12-17:
Sounds good I will start to get a video made.  Do you want me to just put the link in this thread when its done? Or should i email it to you? Considering this is a "inssue" thread and has nothing to do with the issue I'm not sure if you want to keep communicating here? Does not matter to me either way.

yaapu, 2019-12-18:
Hi, you can post it in on [rcgroups]( https://www.google.com/url?sa=t&source=web&rct=j&url=https://www.rcgroups.com/forums/showthread.php%3F3020527-An-Ardupilot-frsky-telemetry-LUA-script-for-the-Horus-X10-X12-and-Taranis-X9D-E-QX7&ved=2ahUKEwj1uOjSq8DmAhXZShUIHWbIA0cQjjgwAHoECAIQAQ&usg=AOvVaw2kAVjFqwmQ5_1hLb0HWINv) or  if you'd like a review before posting you can pm me on rcgroups or ardupilot discuss