yaapu/FrskyTelemetryScript #45: Missing file on GitHub

srhassan71, 2020-01-03:
I cannot find the Yaapu file "taranis_1.8.0_rc1.zip".

yaapu, 2020-01-03:
Hi, why do you need an RC version?

to get the latest release simply download the latest [master](https://github.com/yaapu/FrskyTelemetryScript/archive/master.zip)

srhassan71, 2020-01-03:
Hi Alessandro,

I am following Painless360 instruction and the zip file shows RC version. Is there a new version? 
I also notice that there are three folders I need to copy on my sd card for yaapu to work. What if there are already folders with such names on my SD card with different content? Can I rename yaapu folders?

Appreciate some tips.

Thanks
Riz

yaapu, 2020-01-03:
Hi, did you read the installation [wiki](https://github.com/yaapu/FrskyTelemetryScript/wiki/Installation-on-Taranis-radios) for taranis radios?
Folders should not be renamed, simply copy as explained in the wiki.

srhassan71, 2020-01-03:
Thanks for the link. I guess its better to read than follow videos. I
updated my Taranis from 2.2.3 to 2.2.4, updated SD card and than uploaded
Yaapu. It worked!

Thanks again,
Riz

yaapu, 2020-01-03:
Hi RIz, well yeah videos are fine, but if not maintained they get old very fast.
Good job, I tested the script also with OpenTX 2.3.4.

cheers,

Alex

closed 2020-01-03 by yaapu