yaapu/FrskyTelemetryScript #46: Measure battery capacity in Wh instead of Ah

hsteinhaus, 2020-01-12:
My vehicle uses LiIon batteries, like a lot of more recent builds. These cells start with 4.2V and are flown down to 2.6V. As the energy needed to keep the vehicle in the air is constant, the current rises over the whole flight. The last minute of the flight therefore needs 1.6 times as much amp hours as the first minute, rendering the info about Ah used and Ah left virtually useless. The Wh per minute however, are constant over the flight.

IMHO, there should be an option to switch the display from Ah to Wh completely. The math should be rather trivial - just multiply by recent voltage and change the labels.

BTW, outstanding work, @yaapu! Recently I forgot my GCS at home and didn't even miss it ;-)

yaapu, 2020-01-13:
Hi Holger, thanks for the kind words!

Consumed battery capacity right now it's not a calculated value, it's raw telemetry coming from the FC, that's the reason why the script uses the same unit of measure of the flight controller. No integrals involved and no rounding errors due to slow sampling.
If I were to use Wh I'd need 2 extra menu options, one to enable the feature and one to declare battery nominal voltage necessary to convert battery capacity in mAh from Wh. To calculate consumed battery Wh I'd need to integrate instant power over time which would mean that for instance a telemetry link loss would invalidate calculated battery consumption. Battery current is reported at around 2Hz so there would also be an error in the integral.

I like the idea but I'm not sure I'd be able to provide a dependable information!

yaapu, 2021-07-05:
I just realized after more than a year that you meant instantaneous power not consumed power, in this case it would be easy, just multiply by voltage :-(

closed 2021-09-30 by yaapu