yaapu/FrskyTelemetryScript #48: Crossfire over mavlink

jaguraphg, 2020-02-14:
I think this is more the question rather than a bug. I use Crossfire nano connected to FC over Mavlink protocol. Do I understand correctly that only 'PASSTHROUGH' mode (10) should be used to work this script properly? Or there are any other options to make it working? 

Please point me where I can read more to understand how to make my setup fully working with your nice script using Crossfire RC )

Thanks.

yaapu, 2020-02-19:
Hi sorry for the very late reply! Your only option would be to use mavlink and the MavToPT project but I'm afraid your nano receiver does not have it!

jaguraphg, 2020-02-28:
Hi. Thank you for your reply. Could you please clarify - receiver does not have what? Mavlink? If so then it has. Just trying to understand...

yaapu, 2020-02-28:
if your receiver has mavlink input you can send mavlink from flight controller down to your tx, pair an esp32 running the [MavToPT](https://github.com/zs6buj/MavlinkToPassthru) software via bluetotth and have telemetry on your radio

jaguraphg, 2020-02-29:
May I ask you to read the topic below from the message #650.

https://www.rcgroups.com/forums/showthread.php?3089648-Mavlink-To-FrSky-Passthrough-Converter/page44

The person with nickname 'Target0815' explains that this will not be possible by some reason. I try to understand why but this is still not clear to me. I am pretty technical guy but this is still too difficult to ensure that I am on the right way or not )

yaapu, 2020-02-29:
I do not own a Crossfire system so I can't be sure but if you:
- disable native crossfire sport support
- enable mavlink radio modem
- connect FC serial port to mavlink rx/tx on your receiver
- pair via bluetooth an esp32 running the MavToPT firmware to your TX
- connect the esp32 sport pint to the back of your radio sport input

and either use PPM or SBUS you should be fine.
Since all you need is an esp32 why not give it a try?

jaguraphg, 2020-03-01:
You are right. Already ordered esp32, will try. Thank you.

jaguraphg, 2020-03-01:
Really do not understand why TBS cannot do it. This is probably just the firmware task. Strange guys...

closed 2020-05-26 by yaapu