yaapu/FrskyTelemetryScript #49: Where to download working OpenTX firmware?

DroneMang, 2020-02-25:
Hi,

On a clean install of OpenTX Companion, there seems to be no way to get any firmware other than the newest 2.3.5, which doesn't work correct?  I do not have any prior firmwares saved.  I am assuming that 2.3.4 is the last working one with this telemetry script?  Is there any chance someone can post a working firmware for this telemetry?  Thanks!

-DroneMang

yaapu, 2020-03-01:
Hi, sorry for the late reply.
If you have a taranis you can use [2.3.4](https://www.open-tx.org/2019/12/23/opentx-2.3.4), if you have a Horus or T16 you better use 2.3.5

DroneMang, 2020-03-01:
Hi,

I can't seem to find anywhere to download the actual 2.3.4 firmware. I can
download open tx companion in that version, but their website only allows
me to download the SD card contents and open tx defaults to the newest
firmware version. Maybe there is something simple I am not seeing. Thanks!

DroneMang

stevie22b, 2020-03-13:
Hi DroneMang,
Did you ever solve this? I am having the same issue, cannot find anyway of going back to 2.3.4 on my Taranis X9D Plus. On 2.3.5 i cannot save any changes i make to Yaapu.

Thanks,
Steve.

yaapu, 2020-03-13:
@stevie22b , @DroneMang 
this is the version I am using on my X9D+, I take no responsabilities if you use my build :-)

[opentx-2.3.4-lua-flex-en.zip](https://github.com/yaapu/FrskyTelemetryScript/files/4330557/opentx-2.3.4-lua-flex-en.zip)

build options are: crossfire, flexr9m, multimodule,lua and luac

stevie22b, 2020-03-13:
Thank you Yaapu,

That sounds like it will work for me - i am also using the flex9m. I am very grateful for the link. I cannot see why it should not work on my radio, but i will do plenty of tests before flight.

Thanks once again,

Steve.

stevie22b, 2020-03-13:
All working perfectly!
And i am now able to save config changes to the brilliant Yaapu script. And i have checked through all my models and all seem to be working fine with the firmware you linked to. I am so glad i can now save changes, was a bit soul destroying after finally getting the frsky S.Port link working, only to find i could save any changes to the config. I am very grateful for your help.

Thanks very much once again.

Steve.

skystarUK, 2020-03-19:
I have the same problem, I want to get 2.3.4 to avoid the bug in 2.3.5 .
I can't get Companion to download it.   I read on opentx github that this is the intent not to allow d/l of older versions.

Unfortunately I can't use your file as I am using
Taranis x9d (Not x9d+ )  I'm not sure how different it is, I read on forums x9d+ is very little different from x9d, even so, I don't want to risk crashing my drone.

yaapu, 2020-03-19:
yes 2.2.4 works really good

closed 2020-04-17 by yaapu