yaapu/FrskyTelemetryScript #50: Not able to save the configuration script data

fdapport, 2020-02-25:
Maybe the problem was already known but in the Taranis x9d+ firmware OPENTX 2.3.5 the exit key during the configuration of the parameters of the script go out. Is It a way to create a configuration file like modelname.cfg to configure all the parameters without use the radio to configure it? Coud you have same sample to create it and save to the model/yaapu dirs? Thanks, Franco

yaapu, 2020-02-26:
Hi, yes this is a known bug in OpenTX 2.3.5, and there's nothing I can do work around it :-(
You can create the config file in companion and copy it to the radio but you need an older companion, like 2.3.4!

fdapport, 2020-02-26:
Thanks for your quick answer... Do I need to update the Taranis to 2.3.4 version or I could create the yaapu configuration file on the companion program on the PC? I bought the radio just two weeks ago and I do not have more experience on this new radio. Thanks for the help, Franco

yaapu, 2020-02-26:
You can do both but if you do not need the fixes in 2.35 than you can downgrade, I'm running 2.3.4 on my x9d

fdapport, 2020-02-26:
ok, First I tried to simulate it on 2.3.4 but I was not able to run the script… look like that I have not the full script information on the simulation... I will try to downgrade then.. Did somebody create  bug report on the OPENTX github just to have the fix in the next version of openTX? Thanks , Franco

yaapu, 2020-02-26:
yes, it's already fixed in latest dev version

fdapport, 2020-02-26:
thanks, Franco

yaapu, 2020-02-26:
no problem ;)

closed 2020-02-26 by yaapu