yaapu/FrskyTelemetryScript #51: No values in the telemetry radio screen

fdapport, 2020-03-28:
I have the following question/Issues:
Radio: Taranis X9D+ Firmware: 2.3.5  RX: R9 slim on 868Mhz

When I run the yaapu script   I see all the values send from the F405-Wing but when I go in the telemetry menu of the radio I see only the GPS Coordinate, I see the other sensors (CURR, Volt, ALt, etc) but NOT ANY values.. I am not able to set the special functionality of the vario or the voice features. Is some thinks that is possible to solve? Thanks, Franco

[image]
[image]

fdapport, 2020-03-28:
I have resolved, sorry,, just search again the sensors. It is working.

yaapu, 2020-03-30:
ok great, I'm closing this.

closed 2020-03-30 by yaapu