yaapu/FrskyTelemetryScript #52: How do you setup multiple sites?

tonyyng, 2020-04-18:
[The wiki says:](https://github.com/yaapu/FrskyTelemetryScript/wiki/How-to-configure-the-Mapping-Widget#introduction-to-the-mapping-widget) "For each flying site the pilot will have to download the mapping tiles required to "cover" the area that the widget will display."

Do you:

1. download the tiles for site1, site2, ..., siten
2. process the tiles with mktiles.cmd
3. copy the tiles to the radio

or:

1. download the tiles for site1
2. process the site1 tiles with mktiles.cmd
3. copy the site1 tiles to the radio
4. clear the downloaded tiles for site1
5. download the tiles for site2
6. process the site2 tiles with mktiles.cmd
7. copy the site2 tiles to the radio
8. clear the downloaded tiles for site2
9. download the tiles for site3
10. process the site3 tiles with mktiles.cmd
11. copy the site3 tiles to the radio

yaapu, 2020-04-27:
Hi, it works both ways, feel free to find your own workflow :-)

tonyyng, 2020-04-28:
Great.  Thanks

closed 2020-04-28 by tonyyng