yaapu/FrskyTelemetryScript #53: Taranix x9 lite support?

SecvinPhil, 2020-04-29:
Hello,
I just tried the yaapu scripts on the [Taranix x9 lite ](https://www.frsky-rc.com/product/taranis-x9-lite/) simulator, using Companion 2.3.
However when I try to run the telemetry script "yaapu9", the voice says "yaapu telemetry ready", but the displays only displays "syntax error". 
I kind of expected such an outcome, as the Taranis x9 lite is not listed as a supported Transceiver. 

So my Question: Did I do something wrong, or is the Taranis x9 lite really not supported?
Also: Could you add support for the Taranis x9 Lite :D?

Best regards,
Phil

yaapu, 2020-04-30:
Hi on the X9 Lite you need to run the QX7 version (yaapu7) since the screen is 128x64 just like the QX7 and X-Lite. I did a test in companion and it works.

Please report back so I can add it to the docs, thanks!

SecvinPhil, 2020-05-01:
Okay, I will do. Please give me a couple of days until I set the physical connection up.
Thanks for getting back so quickly!
Do you need any additional documentation appart from works/does not work?

yaapu, 2020-05-01:
once it has been running for a while, would be nice to have the memory stats, long press exit, long press encoder, select statistics, press page and take note of "free mem"

[image]

yaapu, 2020-05-26:
@SecvinPhil any news on this?

SecvinPhil, 2020-05-27:
Sorrz, I havent tested it on the Hardware yet because I have problems with the Telemetry cable. I will try again tomorrow!

SecvinPhil, 2020-06-27:
Hey, I finally got around to manufacturing the telemetry cable. It works beautifully!
I am closing this topic

closed 2020-06-27 by SecvinPhil