yaapu/FrskyTelemetryScript #55: Lost all the sensors if change the model.

fdapport, 2020-05-07:
Not sure if is a yaapu,  Taranis firmware or  APM firmware issue. 

Hardware
Radio Taranis X9D+, XSR Receiver, Betaflight F4 PRO V3

Firmware
TX 2.3.7, RX 2.1.0,  Ardupilot 4.0.5

When I change model on the radio  some time I lost some sensors that I received from the controller via telemetry S.port inverted but I am still see the values  in the yaapy screen display. If I go the telemetry setup menu of the radio I do NOT see the values received.  To solve the issues I cancel all the sensors and I search again it in the telemetry radio menu and it find it again. The negative think of this procedure is that I lost all the configuration, like VARIO or voice setup of the value. Do you know if I known problem or where could it be solve?

yaapu, 2020-05-09:
Yaapu doesn't use sensors.You can remove them all and it will work just fine.
I inject simulated sensor data into OpenTX so the user can use them for vario etc.
I see you are using ACCST 2.1 or ACCESS 2.1 which I never used so can't help much

fdapport, 2020-05-09:
yes,, yaapu works great with any problems...I use ACCST 2.1.....The only problem happen when I try to set voice /alarms on the sensors, like this screen..,,

[image]

some time I do not have any values---ONLY GPS
[image]

Then I cancel the sensors and read again... and the special function get wrong senquence...see the screen
[image]
[image]

fdapport, 2020-05-18:
Resolved: It is NOT a issue related with Yaapu script but is related with save models in Companion script.
See link:
https://github.com/opentx/opentx/issues/7613

closed 2020-05-18 by fdapport