yaapu/FrskyTelemetryScript #56: Request

mogli73, 2020-05-20:
Good day!

I love your script! But I have a request: Can you please include this feature (from athertop) in your script? With this the user can see the position (on the taranis X9D +) of the UAV relative to his position.
Thank you very much!
See red circle:
[image]

yaapu, 2020-05-22:
hi, yes I'd like to add such a view as well, it's planned but haven't worked on it yet.

Alex

elyfranco, 2021-08-08:
Ciao Alex, mi chiamo Franco e sono Italiano vivo a Roma ho letto che anche tu sei Italiano e non puoi capire che piacere di aver trovato una persona che Italiana visto che il mio pessimo inglese non mi aiuta pur utilizzando dei traduttori non è semplice per me capire cose tecniche. Premesso che sono un modellista di vecchia data ma aver scoperto il volo in FPV è come poter avverare la passione di volare, vorrei entrare nel merito e spero che tu possa dedicarmi un poco del tuo tempo al fine di poter far funzionare il LUA nella mia trasmittente Taranis X9D 2019 SE, come F.C. ho Matek 722 wing fireware Inav 2.6.1. Open TX 2-3.13 detto questo ho scaricato il file FrSky Telemetry-Script-Tarnis 1.9.5_beta1, decompresso il file all'interno trovo le tre cartelle immagino che io dovrò utilizzare la cartella Taranis ma anche quella TOOLS? pui indicarmi i successivi passaggi dove inserire e quali file copiare al fine di poter rendere funzionante il Lua sopracitato? Spero di ricevere cortese indicazione, anticipatamente grazie. Buona giornata Franco

mogli73, 2022-02-01:
Good day!
Any news about the enhancement?

yaapu, 2022-02-01:
> Ciao Alex, mi chiamo Franco e sono Italiano vivo a Roma ho letto che anche tu sei Italiano e non puoi capire che piacere aver trovato una persona Italiana visto che il mio pessimo inglese non mi aiuta nonostante l'utilizzo di traduttori non è semplice per me comprendere nozioni tecniche. Premesso che sono modellista di vecchia data, aver scoperto il volo in FPV è come poter avverare la passione di volare, vorrei entrare nel merito e spero che tu possa dedicarmi un poco del tuo tempo al fine di poter rendere funzionale il LUA da te pubblicato nella mia Taranis X9D 2019 SE, come F.C. ho Matek 722 wing fireware Inav 2.6.1. Open TX 2-3.13 detto questo ho scaricato il file FrSky Telemetry-Script-Tarnis 1.9.5_beta1, decompresso il file all'interno trovo le tre cartelle immagino che io dovrò utilizzare la cartella Taranis ma anche quella TOOLS? puoi indicarmi i successivi passaggi e quali file copiare al fine di poter rendere funzionante il Lua sopracitato? Spero di ricevere cortese indicazione, anticipatamente grazie. Buona giornata Franco

Ciao Franco, scusami per l'imbarazzante ritardo nella risposta.
Il mio widget supporta solo Ardupilot non è prevista la compatibilità con INAV/Betaflight

yaapu, 2022-02-01:
> 

nope, I tried but it crashes, no memory left to play with. Might work on X9D 2019

mogli73, 2022-02-01:
Thank you for your quick response!

So you realised the feature but it is not working on a Taranis X9D+?

When I use the script on a Taranis X9D 2019 or Horus it will work already?
Best regards

yaapu, 2022-02-01:
Hi, maps is already supported on Horus by using offline google maps, on X9D I only added the extra screen (empty) and it already crashed for no memory left so I stopped developing.

closed 2024-10-10 by yaapu