yaapu/FrskyTelemetryScript #59: V1.8.0 QX7 OTX2.3.7 Config missing - when long press Menu

WillyZehnder, 2020-06-05:
When I long press Menu, there is a TOOLS list, but "Yaapu Debug" is the only tool from Yaapu. The others are Crossfire, FrSky, Graupner and Spectrum. I searched for *.lua on SD-card, but a "Yaapu Config.lua" or similar is missing.
How can I start the Configuration?

yaapu, 2020-06-11:
Hi, on the Taranis radios like the QX7 you access the config menu by long pressing the [menu] button while on the script main page.

closed 2020-06-11 by yaapu

WillyZehnder, 2020-06-12:
Thank you for the hint Alex!
Maybe you should insert a half-sentence in the README.
configuration menu, long press [MENU] **while on the script main page** on Taranis or long press [SYS] on Horus (refer to this guide)