yaapu/FrskyTelemetryScript #65: RSSI always uses connected 2.4ghz rx, even when RSSI sensor is deleted

ItsMeAubey, 2020-08-10:
I am using https://github.com/zs6buj/MavlinkToPassthru in relay mode. Discovering sensors results in two RSSI sensors, one provided by M2P and one provided by my RXSR. If I delete the RXSR sensor, it is still used as the rssi source in the telemetry script, despite the only existing RSSI sensor being the one provided by M2P (instance 28). Confirmed by wrapping tin foil around my 2.4ghz antenna - rssi drops massively. 

I tried rebooting all hardware and changing models to make sure it wasn't a caching issue. Issue persists.

yaapu, 2020-08-12:
Hi, the script has no direct access to OpenTX rssi sensors, I simply use the getRssi() function, no idea why it would return a value not "there" anymore, doen't make sense at all!

ItsMeAubey, 2020-08-12:
It really is strange.. Perhaps this is a bug in opentx then?

ItsMeAubey, 2020-08-12:
Apparently, the "rssi source" setting in the telemetry tab does more than just the rssi source for alarms.. oops.. configuration issue on my end. I'll add this to the docs for m2p.

closed 2020-08-12 by ItsMeAubey

yaapu, 2020-08-12:
glad you solved it!

ItsMeAubey, 2020-08-12:
Did #66 too to make sure nobody else misses this :)