yaapu/FrskyTelemetryScript #67: Flight Mode Confliction with multiple MAV IDs

mike-bfd, 2020-08-31:
When attaching a Gremsy gimbal to the pixhawk via Mavlink, I noticed that the flight mode constantly changes. When this gimbal is connected to the aircraft, it appears as a second vehicle, with it's own data stream. (In mission planner, you can select the aircraft or the gimbal in the vehicle drop-down next to the connect button). I believe that the script is interpreting both vehicles somehow.

I have attached a video and tlog to hopefully help explain.
[https://drive.google.com/drive/folders/1ZPfU7rXaT2E5CgQCuOD5_RAK1FszSysk?usp=sharing](https://drive.google.com/drive/folders/1ZPfU7rXaT2E5CgQCuOD5_RAK1FszSysk?usp=sharing
)

yaapu, 2020-09-01:
Hi, this is an issue with the MavToPT code on the TX MOD you better ask for help to Eric Stockenstrom here https://github.com/zs6buj/MavlinkToPassthru/issues

mike-bfd, 2020-09-01:
Thanks! Closing here to re-open on Mav2PT

closed 2020-09-01 by mike-bfd

yaapu, 2020-09-01:
Hi Mike, I kind of remember that Eric solved a similar issue with another device acting weird on the mavlink bus