yaapu/FrskyTelemetryScript #76: Yaapu LUA script on Radiomaster TX16S with Crossfire module

ojoojoo, 2020-10-20:
I use OpenTX 2.3.10 and Yaapu LUA 1.8.1 on a Pixhawk 4 mini with a Nano RX receiver from Crossfire.
I installed the script exactly as in the Painless360 video (https://www.youtube.com/watch?v=SAQHowQ3rFM) and the Github manual but I do not get all the available telemetry information in the Yaapu LUA script.
The telemetry information is available on the transmitter, but for example the GPS coordinates are visible in the Yaapu widget, but it says NO GPS in the screen.
[image]

What am I doing wrong, please help!

Cranners47, 2020-10-20:
I have this exact same problem with a Horus X10s, TBS Crossfire and the Pixhawk 2.1.  Whe I go to discover new sensors it lists all the sensors available, but I cannot for the life of me figure out how to get it to display on the telemetry widget!

Kija, 2020-10-20:
Hi guys,

1) there was already a github issue opened about CRSF support in the Yaapu script: https://github.com/yaapu/FrskyTelemetryScript/issues/11
2) the real discussion is happening here https://discuss.ardupilot.org/t/passthrough-telemetry-over-crsf-crossfire/62668

If you haven't read either, chances a very slim that you could get it working.

Cranners47, 2020-10-21:
Ok, this is doing my nut in.  Current setup is Horus X10S Express, running Open TX 2.3.10, using Yaapu 1.8.1.  Tx and Rx are the TBS cross fire Nano RX and full sized TX, F/C is a PixHawk 2.1 Cube Black.

If I configure the Receiver to use Mavlink Tx/Rx on channels 7 & 8, connected to either telemetry port on the Pixhawk, with the SERIALn_PROTOCOL set to 1 (Mavlink1) I can get control of the craft, but yaapu does not pick up the telemetry.  I know it is getting out of the plane and into the controller because I can connect via Bluetooth to the TBS and see the telemetry on Mission Planner.

Following the instructions, I put Yaapu 1.9.0b_0.7 on the SD card of the controller, installed the Version 0.7 Ardu Pilot binary and then tried again.  This time enabling CRSF TX/RX on the receiver and setting the SERIALn_PROTOCOL to 23 (as per the instructions, even though Ardupilot says 23 = RCIN) set RC_OPTIONS to 256, and nothing.  Not even control of the craft.  So tried SERIALn_PROTOCOL = 29 (ardu pilot says 29=Crossfire) and again nothing.  Set it all back to how I had it before, i.e. using Mavlink and it works, but no telemetry on Yaapu...

Where am I going wrong?!

yaapu, 2020-10-23:
@Cranners47 please post your support questions on discuss https://discuss.ardupilot.org/t/passthrough-telemetry-over-crsf-crossfire/62668, RC_OPTIONS needs to be set as current value + 256.
Check that you did not set SERIAL_PROTOCOL 23 on more than 1 uart

yaapu, 2020-10-23:
@ojoojoo please check https://discuss.ardupilot.org/t/passthrough-telemetry-over-crsf-crossfire/62668 you're using the wrong widget version

Cranners47, 2020-10-23:
Forgive my ignorance, but if the RC_OPTIONS is set to 0, I have to manually set it to 256?

yaapu, 2020-10-23:
@Cranners47 well sure, if it's 0 256 is fine!

yaapu, 2020-10-23:
@Cranners47 please connect your crossfire receiver to a UART not to RCIN, then set serial_protocol 23 for that uart, set RC_OPTIONS and you're set

closed 2021-01-01 by yaapu