yaapu/FrskyTelemetryScript #77: The counter resets at 60:00?

Damian650, 2020-11-06:
Hello, please excuse me first, my English is poor, so I am using the translator.
 I was presented with the following problem:
       When I reset the counter, it always goes to 60:00, never 00:00.
       Has it happened to someone else?

My radio is a TX16S Radiomaster.

[image]

yaapu, 2020-11-16:
Sorry for the late reply, I can't reproduce the issue, are you sure you do not have another script/function modifying timer 3 which is used for the flight timer?

Damian650, 2020-11-17:
Hi.
Thank you for taking the time to respond.
Just before reading the answer, I decided to format the SD.
I was installing the scripts one by one, until one of them generated the error, the script is that of "iNav"
So I'm going to choose not to use it.

The Scripts I am using are the following:
[https://github.com/yaapu/FrskyTelemetryScript](url) 
[https://github.com/yaapu/HorusMappingWidget](url)

Y este es el que causaba el error:
[https://github.com/iNavFlight/LuaTelemetry](url)

I am going to reconfigure my radio and when I finish if the problem does not arise again I will give the issue as resolved.

Many thanks.

yaapu, 2020-11-17:
I didn't know lua iNav manipulated timer 3, thanks for reporting back

closed 2020-11-20 by yaapu