yaapu/FrskyTelemetryScript #78: Artificial horizon doing looping

Xelack, 2020-11-18:
Hi Yaapu,

I use your script (V1.8) on QX7 all are working fine but I don't know why when I fly some times I see the artificial horizon doing looping.

Have you an idea on this kind of strange behavior ?

FC PIXHACK V3 with ARDUCOPTER 4.0.3 (TELEM2 at 57000 Baud)
ESP32+SparkFun Logic Level Converter (https://www.sparkfun.com/products/12009) with MAVLINKTOPASSTHRU V2.63.3.

Thanks,

NOTE : LOT OF THANKS FOR YOUR GOOD JOB ;)

BR,

Alexander.

yaapu, 2020-11-20:
Thanks...you're the first to report it....did you check in the ardupilot logs that the IMUs are working fine?

Xelack, 2020-11-28:
Hi Alessandro,

No, all are working fine in arducopter, no issues found in logs.

I believe I found the problem. Your scrypt works fine and it's my radio R9 system which buggy in reception on the telemetry.
So I checked the version of the RX and TX firmware, it was only in v1.2.0 (EU) Access which can cause this kind of reception problem. By reading several forums I saw that people often switch to Flex version so that's what I did with the latest version 1.3.1 RX and 1.3.0 TX. Since no more telemetry or reception problem.
This is another subject, but I am quite disappointed with this material, since the beginning I always have problems with the bind at the power-on, much less now with the 1.3 update but I still have to power-on twice the rx before it binds to me radio. At the first power-up it seems to stay in register mode (red led and green led on), I don't know why because I was already register the rx and I have only one linked to my tx.

Thanks a lot for your support,

Alexandre

closed 2020-11-28 by yaapu