yaapu/FrskyTelemetryScript #80: Hope support synthetic airspeed calculated by flight controller

rxwang04, 2020-12-22:
The synthetic airspeed in arduplane is quite accurate now,it's not necessary to use airspeed sensor just for general flight,pilot should pay more attention to airspeed than ground speed,which means that most people want to display airspeed rather than ground speed,but the frysky library can only send one speed.we set airspeed_type=0 without airspeed sensor,so it can be display only ground speed.
Does it can display synthetic airspeed with airspeed_type=0?

yaapu, 2021-01-01:
not a bad idea but not yet possible unless we change the way the frsky library works, please open an issue on ardupilot and ask for using synthetic airspeed for plane when there's no airspeed sensor

WillyZehnder, 2021-05-26:
@rxwang04 Do you have started an issue on ardupilot? If yes, what number?
I don't use an airspeed-sensor at all and with syncairspeed also soaring is working great. 

So I'm very interested in displaying the synthetic airspeed, too.