yaapu/FrskyTelemetryScript #84: Enhancement: Audio/vocal airspeed alert

cheintz, 2021-01-08:
For plane models especially, it would be useful to receive audio warnings about low (or high) airspeed in a manner similar to what is provided for altitude. More frequent alerts when at low speed and at low altitude would be useful to help avoid stalling while landing.

yaapu, 2021-01-12:
Hi, you can add OpenTX alerts monitoring the GSpd sensor which when an airspeed sensor is present adn enabled reports airspeed and not groundspeed

hyperion11, 2021-11-09:
👍 
also wish this feature :) Like battery alarm

Nerenye, 2022-03-24:
@yaapu i can´t get the Airspeed into Ethos with passthrough telemetry on. is there a way to enable it anyway?

yaapu, 2024-10-10:
this was addressed by "lua sources" in ethos

closed 2024-10-10 by yaapu