yaapu/FrskyTelemetryScript #89: (Minor) Battery voltage calculation comment

rotorman, 2021-01-18:
Please check the following comments

https://github.com/yaapu/FrskyTelemetryScript/blob/82b2245639e209b875b994cc9dd0ae2e9f4e3f0e/HORUS/SOURCES/SRC/WIDGETS/Yaapu/main.lua#L861

https://github.com/yaapu/FrskyTelemetryScript/blob/82b2245639e209b875b994cc9dd0ae2e9f4e3f0e/HORUS/SOURCES/SRC/WIDGETS/Yaapu/main.lua#L871

`52.3 is 0.1...60 is 88` does not make sense IMHO.

I believe correct would be in the following vain:
`for cellcount 1 to 11, telemetry max is 51.1V. For cellcount 12, values 0 to 239 report 51.2V to 75.1V (Vreal = 51.2 + telemetry.batt1volt), values 240 to 511 are reported as 24.0V to 51.1V`

yaapu, 2021-01-19:
Hi that is not how the calculation works, you're missing a couple points.
For voltages > 24V cellcount is not taken into account and telemetry voltage is assumed real voltage. (so 1-12s, voltage > 24)
For voltages < 24 we check the cellcount because users with LiHV might have a full voltage pack > 51.2 and ardupilot right now "wraps" the voltage, so it transmit whatever is in excess of 51.2. If that's the case than apply the conversion 51.2 + transmitted voltage, for this to work the user has to "force" cellcount in the script config for autodetection will fail for sure.

rotorman, 2021-01-19:
Thank you!

If I read the following code snippet:
https://github.com/yaapu/FrskyTelemetryScript/blob/82b2245639e209b875b994cc9dd0ae2e9f4e3f0e/HORUS/SOURCES/SRC/WIDGETS/Yaapu/main.lua#L860-L866

I understand it as:
a) for battery cellcount != 12: real voltage is output, 0 to 51.1V range
b) for battery cellcount ==12: a split at 24.0V is made, where values 0 to 239 output 51.2V to 75.1V and values 240 to 511 output 24.0V to 51.1V.
(and the comment on line 863 should read: 1 is 51.3V, 88 is 60.0V)

Did I get that correctly?
---------
And ArduPilot already outputs over MAVLink then values higher than 51999 = 51.999V, but wraps them from 0, did I get that correctly? Meaning a value of 1 on MAVLink can mean 1mV or 52.001V ?

closed 2021-04-18 by yaapu