yaapu/FrskyTelemetryScript #91: No telemetry data

marcelking, 2021-01-30:
can someone help?
running 1-9-3 beta its telling me no telemetry data even though the gps for
ordinates are there , i am running crossfire which i have enabled in the configuration do i need to set the crossfire receiver to any settings? i can't find any info on this.
the tbs receiver is hooked into a nav enabled fight controller,.

As a note the inav telemetry screen is receiving all data and working correctly
many thanks for any help

yaapu, 2021-02-01:
Hi, are you running ardupilot? If so which version?

marcelking, 2021-02-01:
Hello I am not running ardupilot I am running inav on a matek flight
controller
Regards

yaapu, 2021-02-01:
Hi my scripts only support ArduPilot :-(

closed 2021-02-01 by yaapu

marcelking, 2021-02-01:
Ah sorry I didn’t know that good job I run a Pixhawk as well then 😁
brilliant scripts though
Regards
As a after thought the inav can put Mavlink protocol through?

yaapu, 2021-02-01:
Perhaps the INAVLua script is a better alternative, you can pair it with my other widget for GPS mapping support https://github.com/yaapu/HorusMappingWidget

Not sure how good iNav would work using mavlink, anyway you should add a mavlink decoder and use the MavToPT project

marcelking, 2021-02-01:
Hello again
Out of interest for you the maps work you can see the aircraft move over
the map so it’s still excellent 😁

yaapu, 2021-02-01:
yes, basic map functionality only requires an OpenTX  "GPS" sensor which is the only one provided by pretty much any autopilot

marcelking, 2021-02-01:
That’s brilliant thank you I can get the telemetry from the inav script and
see where I am on the map with your script 😁
Regards