yaapu/FrskyTelemetryScript #92: gmap catcher

Eg681, 2021-02-07:
Ive been fighting this for 2 days and I see other people having the same problem as me, I downloaded the Gmap catcher from github no problem but when map opens up I have only standard view no satellite or terrain view which makes this useless to fly with since there is not enough detail in standard view of map. Is there a work around for this? will this issue be addressed?

yaapu, 2021-02-09:
Hi, there's no issue, simply some map providers do allow  map downloading, try with Virtual Earth (Bing)

[image]

also consider setting virtual earth as download provider

[image]

all this is very well explained in the GMapCacther documentation.

Please note that I'm not the author of gmapcacther.

closed 2021-02-09 by yaapu

yaapu, 2021-02-09:
...gmapcatcher has been updated to version 0.8.2.1 to support latest nokia API, I updated my custom version as well, now nokia maps (default) is working again

heldersepu, 2021-02-09:
Hello Hello...
I'm the owner of [GMapCatcher](https://github.com/heldersepu/GMapCatcher) and Yes I got a few reports that it was "not working" on any combination of map servers, quite shocking for me because on my tests it worked fine ...
Turns out it was just the Nokia server api...

@yaapu I see you do have a custom version:
https://github.com/yaapu/FrskyTelemetryScript/wiki/How-to-configure-the-Mapping-Widget
maybe update/sync that with latest

yaapu, 2021-02-09:
@heldersepu  yep, sync done, thanks!

heldersepu, 2021-02-09:
@yaapu that is awesome! 

Was looking at your fork and I think you can take advantage of less map servers (hide_map_servers):
https://github.com/yaapu/GMapCatcher/blob/master/gmapcatcher/mapConf.py#L152
For your audience I think most of the maps offered by GMapCatcher are not really of any use, best to just have those you really use and hide everything else