yaapu/FrskyTelemetryScript #93: Feature Request: Roll and Pitch setTelemetryValue

0815Creeper, 2021-02-10:
it would be nice if the script could output the roll and pitch angle (possibly in degrees) to otx, so that it can be logged/used by otx just as altitude, speed, voltage, GPS etc.
Would this be possible without too much work?

yaapu, 2021-07-04:
Hi, 
you do not specify your radio model, generally speaking it would be easy but on Taranis radios it would use memory which I rather save for future features, you can try to add it yourself by uncommenting two lines, search the sources for setSensorValues() and uncomment following 2 new lines, one for roll and one for pitch, something like

  setTelemetryValue(0x070F, 0, 0, telemetry.roll, 20 , 0 , "ROLL")
  setTelemetryValue(0x071F, 0, 0, telemetry.pitch, 20 , 0 , "PTCH")

0815Creeper, 2021-07-04:
Yea, thanks. I could not find any instructions on how the preprocessor/compiling works. As this does not seem to be necessary, I already got it working. Thanks

closed 2021-07-04 by 0815Creeper