yaapu/FrskyTelemetryScript #94: Switch to show of min/max values don't working on Horus widget

wizardrnd, 2021-02-12:
I am using the latest script version 1.9.3 beta4 on my Radiomaster TX16S, and try to switch to show min/max values by short press of [SYS] as described this: https://github.com/yaapu/FrskyTelemetryScript/wiki/Features#advanced-features , but it don't working.

Of course, I can add another page with this widget, set the widget settings to "page 3" and switch between these pages, but it's not very convenient.

yaapu, 2021-02-12:
Hi, wiki needs to be updated sorry, it used to work when script was not a widget. When run as a widget there's no way to interact with the buttons so no way to detect long press of [SYS]

closed 2021-02-12 by yaapu

wizardrnd, 2021-02-17:
And what about switch between the current and min/max indicators by using emulated wheel channel? 
This channel is not used on script 1st page