yaapu/FrskyTelemetryScript #96: Feature Request: Show Distance to WP

kkarl999, 2021-02-23:
Salve Alessandro, 

spero che tutto vada bene con te. 

It ist possible to change or add a view Distance to Waypoint? This is a nice feature for Rover / Boats. 

Thank you , Mille Grazie 
Daniele

yaapu, 2021-02-23:
Hi,
WP distance is already there but it requires MavToPT or a board that supports scripting, what is your setup?

kkarl999, 2021-02-24:
Hi Alessandro, 
oh cool, i have searched but nothing to find about MavToPT. 
My Setup is a Rc Baitboat with Radiolink Pixhawk 2.4.8 and two Radiolink Se100 GPS. 
The Radio is a Radiomaster tx16S with Radiomaster R168 Receiver. 
OpenTx 2.3.11 Firmware 

I Plan to Update my Setup with a TBS Crossfire Tx with a TBS Crossfire Diversity RX Receiver. 

For Boat Users is the WP Distance very Important. 

Mille Grazie, Thank you 
Kind Regards 
Daniele

yaapu, 2021-03-04:
ok, I need to write up an example in the docs

yaapu, 2021-07-04:
Hi Daniele, were you able to set it up?

kkarl999, 2021-07-04:
Hi Alex, 
thank you for your reply, 
no i didn't get it, i didn't understand from where to where i should copy the files. :-( didn't have much time to try.
Would you like to describe it to me in more detail? then I'll try again.

yaapu, 2022-11-26:
available in latest Horus version

closed 2022-11-26 by yaapu