yaapu/FrskyTelemetryScript #103: Incorrect Flight Mode Audio

Arktix13, 2021-03-17:
I'm getting incorrect audio messages when I change flight modes on a TX16S controller. The quadcopter I'm connecting to is set up with Manual, Altitude, and Position modes. Yaapu Telemetry correctly says "Manual Flight Mode" when I'm in Manual, but when I'm in Altitude it says "Stabilize Flight Mode". Even more confusing, when I go into Position it says "Acro Flight Mode". How does the script determine the flight modes? Trying to figure out why it's getting the modes wrong and how to correct it.

Arktix13, 2021-03-17:
Update: Similar issue on Taranis X9D+ SE 2019. I didn't write down what it was saying for Altitude and Position modes, but it's reading them wrong. I saw there is a setting to enable/disable reading PX4 flight modes, but toggling it off doesn't seem to turn the audio readouts off.

Arktix13, 2021-04-06:
How does the script determine which flight mode audio .wav file to play?

yaapu, 2021-04-18:
What is your exact setup? Are you running the PX4 stack or ardupilot?
If PX4 which version of MavToPT are you running?
The "enable PX4 flightmodes" option has to be enabled but works only with PX4 + MavToPT

Arktix13, 2021-04-18:
So the aircraft is running PX4 and I have your script loaded onto a TX16S. I wasn’t aware MAVToPT was required for PX4. I’ll try installing that and see if it fixes the audio issues. Which version of MAVToPT do you recommend?

yaapu, 2021-04-20:
RFD900 has built in MavToPT firmware, that's why it's working, mostly. Flight modes mapping is mavlink->txmod->frsky->widget.
I personally tested PX4 flight modes support several times but I used standard MavToPT firmware and never tested first hand a TXMOD, what I'm saying is I'm sure the mapping is correct IF using MavToPT

Arktix13, 2021-04-29:
So I'm still having issues with incorrect flight mode audio. I've been in touch with RF Design about MavToPT on the TXMOD. They confirmed that they were able to reproduce the issue. The setup we're both using is a TXMOD in a RadioMaster TX16S controller and a PixHawk 2.1 black cube running PX4. RF Design noticed that the MavToPT function for PX4 modes does not match the flight mode function in the Yaapu script. This is probably why we're getting incorrect audio messages. See below.

MavToPT:
uint8_t PX4FlightModeNum(uint8_t main, uint8_t sub) {
 
  switch(main) {
    
    case 1:
      return 0;  // MANUAL 
    case 2:
      return 1;  // ALTITUDE       
    case 3:
      return 2;  // POSCTL      
    case 4:
      switch(sub) {
        case 1:
          return 12;  // AUTO READY
        case 2:
          return 13;  // AUTO TAKEOFF 
        case 3:
          return 14;  // AUTO LOITER  
        case 4:
          return 15;  // AUTO MISSION 
        case 5:
          return 16;  // AUTO RTL 
        case 6:
          return 17;  // AUTO LAND 
        case 7:
          return 18;  //  AUTO RTGS 
        case 8:
          return 19;  // AUTO FOLLOW ME 
        case 9:
          return 20;  //  AUTO PRECLAND 
        default:
          return 31;  //  AUTO UNKNOWN   
      } 
      
    case 5:
      return 3;  //  ACRO
    case 6:
      return 4;  //  OFFBOARD        
    case 7:
      return 5;  //  STABILIZED
    case 8:
      return 6;  //  RATTITUDE        
    case 9:
      return 7;  //  SIMPLE 
    default:
      return 11;  //  UNKNOWN                                        
   }
}
 

Yaapu Script:

local flightModes = {}
--
  | -- plane flight modes
  | flightModes[0]=""
  | flightModes[1]="Manual"
  | flightModes[2]="AltCtl" --px4   specific
  | flightModes[3]="PosCtl" --px4   specific
  | flightModes[4]="Ready" --px4   specific
  | flightModes[5]="Takeoff" --px4   specific
  | flightModes[6]="Loiter"
  | flightModes[7]="Mission" --px4   specific
  | flightModes[8]="RTL"
  | flightModes[9]="Land"
  | flightModes[10]="RTGS" --px4   specific
  | flightModes[11]="Follow"
  | flightModes[12]="PrecLand"   --px4 specific
  | flightModes[13]=""
  | flightModes[14]="Acro"
  | flightModes[15]="OffBoard"   --px4 specific
  | flightModes[16]="Stabilize"
  | flightModes[17]="RAttitude" --px4   specific
  | flightModes[18]="Simple" --px4   specific
  | flightModes[19]=""
  | flightModes[20]=""
  | flightModes[21]=""
  | flightModes[22]=""
  | flightModes[23]=""
  | --
  | return {flightModes=flightModes}

yaapu, 2021-04-29:
hi, that was a bug corrected a while ago, did you test latest  1.9.x beta version?

PX4 modes in version 1.9.x are:

```
local flightModes = {}
-- plane flight modes
flightModes[0]  = "Manual"
flightModes[1]  = "AltCtl"     --px4 specific
flightModes[2]  = "PosCtl"     --px4 specific
flightModes[3]  = "Acro"
flightModes[4]  = "OffBoard"   --px4 specific
flightModes[5]  = "Stabilize"
flightModes[6]  = "RAttitude"  --px4 specific
flightModes[7]  = "Simple"     --px4 specific
flightModes[8]  = ""
flightModes[9]  = ""
flightModes[10] = ""
flightModes[11] = ""
flightModes[12] = "Ready"     --px4 specific
flightModes[13] = "Takeoff"   --px4 specific
flightModes[14] = "Loiter"
flightModes[15] = "Mission"   --px4 specific
flightModes[16] = "RTL"
flightModes[17] = "Land"
flightModes[18] = ""
flightModes[19] = "Follow"
flightModes[20] = "PrecLand"  --px4 specific
flightModes[21] = ""
flightModes[22] = ""
flightModes[23] = ""
flightModes[24] = ""
flightModes[25] = ""
flightModes[26] = ""
flightModes[27] = ""
flightModes[28] = ""
flightModes[29] = ""
flightModes[30] = ""
flightModes[31] = "Unknown"

```

Arktix13, 2021-04-29:
I have been using HORUS 1.8.1 since that is the latest stable version. Which do you recommend I use?

yaapu, 2021-04-29:
got for latest https://github.com/yaapu/FrskyTelemetryScript/releases/tag/Horus-Widget-1.9.3-beta4

Arktix13, 2021-05-05:
I tested out 1.9.3-beta4 today and that did indeed resolve the incorrect flight mode audio issue. Thanks for your help with this!

closed 2021-05-10 by yaapu