yaapu/FrskyTelemetryScript #104: No mavlink messages with RadiomasterT16s Crossfire v2

Alexibexi38, 2021-03-18:
Hello first i have to say, that your script is awesome! Great work!
But sadly I can't get it to work on my t16s.
I see the sensors like RxBt or GAlt and so on, but there are no mavlink messages.
So is my setup supported? And if not would it be a new Feature soon?
Thanks a lot!

yaapu, 2021-03-21:
what is your setup?ardupilot version, crossfire version? opentx version and my widget version?

Arktix13, 2021-03-22:
@Alexibexi38 make sure you are using MAVlink v1 everywhere (on the Crossfire, in any GCS software you're using, etc.). In my experience, the Crossfire doesn't do well with MAVlink v2. Still, you may not be able to get this to work with a Crossfire. It's designed primarily for a low latency FPV connection and has limited bandwidth for transmitting telemetry data. I've had issues of this sort with Crossfire. I've found the RFD 900 modems to be much better.

yaapu, 2021-04-18:
Hi @Alexibexi38 you need to setup ArduPilot to use the CRSF protocol by SERIAL_PROTOCOL = 23 and enable custom telemetry for crossfire with RC_OPTION+= 256 (add 256 to your current RC_OPTIONS value), you also need ardupilot 4.1-dev
Much more infos here https://discuss.ardupilot.org/t/passthrough-telemetry-over-crsf-crossfire/62668/652

closed 2021-05-05 by yaapu