yaapu/FrskyTelemetryScript #109: gMapCatcher not getting tiles

jasc76, 2021-05-05:
Your custom version seems to be incompatible with map servers. I tried all and only get very sparse tiles

yaapu, 2021-05-10:
Hi, can youi provide a couple locations that do not download, I need to reproduce the issue to debug it, thanks

jasc76, 2021-05-10:
I set Hanau, Germany and try to download north of the city

wolfeeldiver, 2021-05-28:
Love the work you are doing yaapu, very much appreciated.
Seems the modified gmapCatcher 0.8.1.0 does not connect properlty to a map source on min Win 10pro machine to the source and fetch tiles. 
However the latest unmodified 0.8.2.1 version does correctly connect and download tiles. AS if this downloading issue was fixed in 0.8.2.1.   However the problem with this version is that there is no integrated tile resizing to 100x100, it only downloads 256x256 and there is no \100x100 directory made either. 
I was trying at Bellingham, WA USA.

yaapu, 2021-05-28:
Hi, did you try my custom 0.8.2.1 version?
https://github.com/yaapu/GMapCatcher/releases/tag/yaapu-0.8.2.1

wolfeeldiver, 2021-05-28:
Hello. Thank you for the reply. No..I did not try that custom version.I must have missed that one. Thank you I shall give it a try.Greetings from Bellingham WA USA.Thank you.MikeSent from my Verizon, Samsung Galaxy smartphone

https://github.com/yaapu/GMapCatcher/releases/tag/yaapu-0.8.2.1

wolfeeldiver, 2021-05-28:
Hello again,. 
I downloaded your customized version 0.8.2.1 and installed that, and now the gmapcatcher successfully downloads maps from online sources, and also creates the 100x100 tiles.  Thank you.

closed 2021-05-28 by yaapu