yaapu/FrskyTelemetryScript #110: Show the sonar(кфтпу value

AlksSAV, 2021-05-12:


AlksSAV, 2021-05-12:
Is it possible to add a display of values from sonar or plot a graph?
 I'm using sonar readings ( https://ardupilot.org/rover/docs/common-kogger-sonar.html?highlight=kogger) in the Mission Plan. 
My equipment: Pixhawk 2.4, Radiomaster T16S,  Frsky X4R, Kogger sonar. Ardupilot Rover 4.0

yaapu, 2021-05-25:
Hi, the frsky library already supports sending rangefinder data pointing down i.e. RNGFND1_ORIENT = 25, so by enabling rangefidner support in my script you should get the readings.
Plotting it may require a custom panel, I've done some testing by plotting altitude on a rolling graph of 20 samples, it works but to embed it in the widget the number of samples has to be kept low, what are your requirements?

yaapu, 2021-05-28:
this is what I can do now, this are 50 samples, not sure if it will work on a real radio
[image]

AlksSAV, 2021-06-01:
> Hi, the frsky library already supports sending rangefinder data pointing down i.e. RNGFND1_ORIENT = 25, so by enabling rangefidner support in my script (use the config menu) you should get the readings.
> Plotting it may require a custom panel, I've done some testing by plotting altitude on a rolling graph of 20 samples, it works but to embed it in the widget the number of samples has to be kept low, what are your requirements?

Thank you for your work. What do you mean by the configuration menu? As I understand it, first you need the sensor to be in the list, but I do not have it( it is not found in the search). Attached is a photo with a list of sensors. To add it manually, you need as I understand the ID number, but I do not know it for RangeFinder. At the same time, the sonar telemetry data is transmitted to the MP.

AlksSAV, 2021-06-01:
[image]
My list sensors

AlksSAV, 2021-06-01:
> this is what I can do now, this are 50 samples, not sure if it will work on a real radio
> [image]

It looks great, it's more than enough for me.

yaapu, 2021-06-18:
Hi try this version please, in the config menu there's an option to select the plot sources,
[image]

it will enable plotting those values in a dedicated screen

[image]

As a last step assign page 6 to a widget screen the same way as 2 is used for message history, see [here](https://github.com/yaapu/FrskyTelemetryScript/wiki/How-to-run-the-Yaapu-script-on-X10-and-X12-as-a-Widget) for more info.

[SD.ZIP](https://github.com/yaapu/FrskyTelemetryScript/files/6678469/SD.ZIP)

AlksSAV, 2021-06-24:
> Hi try this version please, in the config menu there's an option to select the plot sources,
> [image]
>
> it will enable plotting those values in a dedicated screen
>
> [image]
>
> As a last step assign page [...]

I installed the script version for sending, almost everything was successful. The results are shown in the photo below. 
The test was carried out by immersing and removing the boat from the small barrel. 
Scaling does not work in the graph( that is, the value is not clear, and the instantaneous value is not output( the RSSI value was output, for example).

[image]

[image]

yaapu, 2021-06-25:
mhmm, I sample the value every 1 sec, I will check but I had the rangefinder working, thanks for testing

AlksSAV, 2021-06-26:
Below is a short video with testing. In this video, I immerse and remove the sonar from a container of water. If possible, please add the ability to display only one chart( only for example Sonar, without Home distance or other items).

https://user-images.githubusercontent.com/43313549/123504091-e2308580-d65f-11eb-8134-2818cef6278a.mp4

yaapu, 2021-07-04:
Hi, can you try this version
https://github.com/yaapu/FrskyTelemetryScript/archive/refs/heads/dev.zip

I split rangefinder which uses meters from sonar which uses centimeters, they can be selected individually.
Let me know if this fixes the scale issue

AlksSAV, 2021-08-14:
> Hi, can you try this version
> 
> https://github.com/yaapu/FrskyTelemetryScript/archive/refs/heads/dev.zip
> 
> 
> 
> I split rangefinder which uses meters from sonar which uses centimeters, they can be selected individually.
> 
> Let me know if this fixes the scale issue

I tried this build, the result is in the photo. The value is not displayed, and at what scale the values are.[image]

yaapu, 2021-09-30:
version 1.9.5
[image]

closed 2021-09-30 by yaapu