yaapu/FrskyTelemetryScript #113: GMS coordinate conveersion

Guili123, 2021-05-31:
Hi,

I am using an HGLRC F7 Flightcontroller, with ExpressLRS and Radiomaster TX16s.
Betaflight 4.3 reports DD (decimal degress) coordinates, and gets a fix.
But on TX16s I receive GPS telemtry coordinates in DMS Format.

It seems the script expects GPS DD format and therefore try to show tiles which are far away. 
These tiles was not downloaded before so nothing is shown.
Is it possible to get an option to choose which GPS coordinates format is used?

Is this a know issue and can it otherwise fixed?

yaapu, 2021-05-31:
hi, there's no such thning as coordinate system for OpenTX sensors, what you describe is the rendering not the coordinates, try plotting the values in a telemetry page outside of my script and you'll see what I mean

Guili123, 2021-05-31:
Yeashjust discovered it in parallel.... My problem seem to be that the DMS format is missing the "xxx seconds.
So I get only 50°01 08°01E in the sensors telemtry tab, which is several hundrets meter away from the real GPS coordingates shown in the OSD.
Is there any setting which can be configured to get the "xxx seconds via Telemtry?
^^^I know not script related aanymore, but probably someone know an answer?

yaapu, 2021-05-31:
in my widget config menu you can choose to use decimal

yaapu, 2021-05-31:
do not use the sensor telemetry tab but add a telemetry page with the sensor display, you'll have full resolution

Guili123, 2021-05-31:
Thanks this was helpful. You are right it is a question of rendering , the Sensor telemtry tab shows the full value.
So I must have done an error working with the guide with ardupilot and the converter and copying it to the sd card, because the script show me no map data.

closed 2021-06-18 by yaapu