yaapu/FrskyTelemetryScript #116: Horus Widget 1.9.4 beta1 shows only disabled

twistedwings, 2021-06-11:
after upgrading from 1.8.1 the widget shows only text "disabled".

Radiomaster TX16S
OpenTX 2.3.11

madjan06, 2021-06-15:
I have the same problem.

HORUS X10S
OpenTX 2.3.1
Arduplane 4.1Beta
Crossfire 3.78

In the telemetry-widget-settings I get the following error:

refresh():?:0: attempt to perform arithmetic on field 'terrainLastData' (a nil value)

yak-54, 2021-06-15:
Try update opentx to 2.3.11 ?
opentx 2.3.1 is over 3 years old Firmware for that radio a lot of LUA changes

madjan06, 2021-06-16:
It is 2.3.10, sorry I missed a zero, it was a little bit late yesterday, but I try to update to 2.3.12 today. 
Maybe that helps

depeurvince, 2021-06-16:
I got the same problem.

Radiomaster TX16S
OpenTX 2.3.12
Arduplane 4.1Beta
Crossfire 4.11

In the telemetry-widget-settings I get the following error:

refresh():?:0: attempt to perform arithmetic on field 'terrainLastData' (a nil value)

If I delete the widget and set it again, it works but at next startup it crashes again..

Any ideas?

yaapu, 2021-06-18:
guys please try to download this dev version and let me know if it fixes the issue, simply overwrite what you have installed now thanks :-)

[SD.ZIP](https://github.com/yaapu/FrskyTelemetryScript/files/6678466/SD.ZIP)

depeurvince, 2021-06-18:
Thanks a lot, eager to try it! Unfortunately I won't be able to try before Monday. 
Wish you a nice weekend!

madjan06, 2021-06-19:
Hi,
I have tested it, but I do not know if it is behaving properly.

First try it did not work, disconnected everything from power, repowered the Horus and saw the telemtry widget, no error.
After that I powered my fc and got telemtry data and the widget worked fine. 
Then I disconnected the fc from power, the widget still worked, but as soon as I repower the fc the widget runs into the same error as before and I have to disable the widget and reenable it or restart the horus to get it working again.

If you need more Information I try my best to provide them, thanks a lot for your work ;-)

yaapu, 2021-06-19:
@madjan06 do you get

refresh():?:0: attempt to perform arithmetic on field 'terrainLastData' (a nil value)

madjan06, 2021-06-19:
> @madjan06 do you get
> 
> refresh():?:0: attempt to perform arithmetic on field 'terrainLastData' (a nil value)

Yes, exactly

depeurvince, 2021-06-23:
Hi Alex, 

I tried the dev version that you posted, the problem remain the same.
The widget crashes at every startup, right after the FC initialisation and most probably randomly during the flight.
Setting the widget again in the telemetry page makes it working until the next crash.

yaapu, 2021-06-23:
I can't crash it, which OpenTX version and which ArduPilot version?

yaapu, 2021-06-23:
frsky or crossfire?

madjan06, 2021-06-24:
@yaapu 

@depeurvince wrote a few posts above that he uses the following:
Radiomaster TX16S
OpenTX 2.3.12
Arduplane 4.1Beta
Crossfire 4.11

The only difference in my setup is, that I use a Horus x10s with OpenTX 2.3.10.

depeurvince, 2021-06-24:
Yes exactly! 
It crashed with openTX 2.3.11 as well.
I did few flights yesterday, it crashes during the flight. 
It seems that it happens when there is a change between RF mode 1 and 2 but difficult to say while piloting..

yaapu, 2021-06-24:
ok, guys sorry for not reading the whole thread :-)

depeurvince, 2021-06-24:
No worries 😉

yaapu, 2021-06-24:
I've found the bug :-) fix is coming

yaapu, 2021-06-24:
please try this version and thanks for the feedback!

[SD.ZIP](https://github.com/yaapu/FrskyTelemetryScript/files/6711950/SD.ZIP)

depeurvince, 2021-06-25:
Great, congratulation, it seems to work!
At least no more crashes at startup, will test in flight asap and give you the feedback.

Many thanks!

madjan06, 2021-06-25:
Seems to be working for me also, great work @yaapu 😉

yaapu, 2021-06-25:
Great!

DarkSoulRL, 2021-06-27:
Yes! ty, working for me too.

yaapu, 2021-06-27:
Ok I'll close it, thanks everynody

closed 2021-06-27 by yaapu

padcom, 2021-06-28:
Could you update the repository with the latest sources? Maybe create a branch with the latest sources? It is really hard to check the latest code and what kind of fixes you do if the source is not published. Only `dev` and `master` branches are visible, but those are ages old. 

Pretty please?

yaapu, 2021-06-28:
Yes, the repo is a bit behind, master a lot, dev 2 months. I'll see what I can do

padcom, 2021-06-29:
@yaapu Do you think I should open the same issue for Taranis? I do get the same error, pretty much 100% of the time when I open the telemetry screen. Nothing works.

yaapu, 2021-06-29:
@padcom it's the same issue, fixed on my dev branch, need some time to publish it, sorry

padcom, 2021-06-29:
Btw, I checked the SD.zip on my Jumper T18 (otx 2.3.11) and this is what I get:

https://www.youtube.com/watch?v=s8n394-Xrks

When I try to open one of the telemetry screens I can see a brief flash as if the app would start and then the text "disabled" shows in the left-top corner of the screen. I am not receiving any data besides the last message about RC control being out.

yaapu, 2021-07-04:
> Could you update the repository with the latest sources? Maybe create a branch with the latest sources? It is really hard to check the latest code and what kind of fixes you do if the source is not published. Only `dev` and `master` branches are visible, but those are ages old.
> 
> Pretty please?

Hi repository updated, check the dev branch

yaapu, 2021-07-04:
> Btw, I checked the SD.zip on my Jumper T18 (otx 2.3.11) and this is what I get:
>
> https://www.youtube.com/watch?v=s8n394-Xrks
>
> When I try to open one of the telemetry screens I can see a brief flash [...]

better use the discuss forum I don't think this is a bug but a rather a config issue, I tried my script on the T18 in the sim and worked just fine

padcom, 2021-07-04:
@yaapu Configuration issue?! Are quite serious? An out-of-the-box Jumper T18 with just added your telemetry script with no configuration done whatsoever (besides enabling the CRSF support) can have a configuration issue?

padcom, 2021-07-04:
After reinstalling everything from the `dev` branch the problem is gone. Thanks. Will see if the bug will resurface.

yaapu, 2021-07-04:
>@yaapu Configuration issue?! Are quite serious? An out-of-the-box Jumper T18 with just added your telemetry script with no configuration done whatsoever (besides enabling the CRSF support) can have a configuration issue?

as strange as it may sound too many times (hundreds) it turned out to be some sort of user config issue, as trivial as forgetting to eject the SD and corrupting some files, so my first advice is always double check everything especially because such a big bug as not working on a clean install would have raised so many issue request that in no way it would have gone unnoticed, anyway glad you had it working